ChampSim/ChampSim #2: Deadlock on STLB Merge

ST-mrl, 2018-06-08:
Suggest inserting additional check into PACKET_QUEUE::check_queue lines 20, 42, 62 to verify that entry[i].instruction == packet->instruction. 

This is a workaround for an issue where requests simultaneously reach STLB from the ITLB and DTLB for the same address. They get merged, and then when the request is filled, only the first parent is notified, starving the other and causing deadlock.

ChampSim, 2018-10-27:
This one is a good catch. We'll investigate this issue further. If you have an example trace that shows this deadlock issue, please share it with us so that we can resolve this issue more quickly.

ngober, 2021-02-26:
I'm closing this because it was probably fixed in #87. If it was not, let me know and we can look into it again.

closed 2021-02-26 by ngober