ChampSim/ChampSim #4: Traces Clarification

sarabjeetsingh007, 2018-07-09:
Hi,
Are the traces generated by the provided pintool, for physical addresses or virtual?

ST-mrl, 2018-07-27:
Virtual address. E.g. look at ooo_cpu.cc lines 186/187 -- both destination_memory[i] (physical) and destination_virtual_address[i] (virtual) are set to the same value read in from the trace. 

The rest of the TLB logic is in there (e.g., see line 1562), so if you extend the pin tool to include fields for physical address, and update the lines in handle_branch that create a new arch_instruction so that physical and virtual hold different values, it looks like the code should work out properly...

closed 2018-07-28 by sarabjeetsingh007

Shafayet-Anik, 2021-11-12:
Hi, 
Can you tell me how to read CRC2 traces? trace files may be in binary format, I tried with python, but still found not readable!!