ChampSim/ChampSim #5: Cache coherency and inclusion

majid169, 2018-07-11:
Hey,
1- ChampSIm implements a non-inclusive cache hierarchy. right?
2- How does ChampSim maintain the coherency? I could not find anything regarding the coherency.

ST-mrl, 2018-07-27:
Values are filled to all caches up to their fill level (e.g. see cache.cc line 137 where the upper level cache is handed a value being filled into a lower level cache), however when a victim is found in a lower level cache, higher level caches aren't notified to maintain coherency (e.g. search for 'upper_level' in cache.cc to see every time a cache sends a message to a cache at a higher level).

closed 2020-06-21 by ngober

julax92, 2020-10-20:
I have one question about this ST-mrl, if you do not have coherence of values between L1 and L2, how do you simulate coherence for multicore configurations? Or coherence between multiple core instances it is included?

GinoAC, 2020-10-20:
For multicore simulation each core is assumed to have its own independent trace with no sharing enabled so there is no coherence policy enforced between cores.

julax92, 2020-10-21:
Hello, thanks for your answer. But a couple of more questions came out in my head. First, the model that you comment is for working with threads of different process? Second, can I run a multicore program using this methodology? Let me give more details. I have a Sparse Matrix Vector multiplcation, and this program can be executed without coherency (partioning inputs and outputs). Is it possible to execute this program?

ngober, 2020-10-22:
It is certainly possible to run a multicore program, though there is currently no mechanism for synchronization. If the inputs and outputs are partitioned, that is one less hurdle to clear. You will need to trace your program, possibly using our provided PIN tool, then run ChampSim in its multicore configuration.

In any case, a closed issue from 2018 is not the place to be having this discussion. I recommend you ask on our mailing list (https://groups.google.com/g/champsim?pli=1), where you are more likely to get an answer from someone who has tried this before.