ChampSim/ChampSim #12: Putting efforts in a proper documentation?

itisntalex, 2019-03-29:
For this issue, I'll just pretend to be 6 months back in time when started my Ph.D degree.

I had to put my hands on*ChampSim* and god it was hard! :face_with_head_bandage: 
This project is in a nutshell quite good. It's simple, efficient, it does what you expect from it. But, the main issue with it is to put your hands on it. At first working with cache replacement policies was a nightmare, not understanding the chronology of the execution and so on.

I would suggest to put some effort in writing a proper documentation for this project. In humble opinion, reading a documentation could save sometime in comparison to going into the depth of the simulator's code. 

I'm waiting for your opinions. :+1: :fr:

ChampSim, 2019-04-05:
Yes, we've been asked for a simple manual for ChampSim simulator. ChampSim developers are working on this but it is likely to be released after DPC-3.

ngober, 2020-04-16:
I think it's worth bumping this thread. It has been a while since DPC3, and we're now in the middle of IPC1. Could we use the GitHub wiki for this, so that it's open and anyone can contribute?

itisntalex, 2020-04-16:
> I think it's worth bumping this thread. It has been a while since DPC3, and we're now in the middle of IPC1. Could we use the GitHub wiki for this, so that it's open and anyone can contribute?

I believe anything such as a documentation, whatever the form of it could be of great values for people and community. 😄

akshatjoshi2001, 2022-11-04:
Hi all. Is there any update on this? When can we expect the documentation to be out?

ngober, 2022-11-04:
There are a few articles on the wiki, a brief tutorial in the README, and there is a recently released paper. Is there an area you feel that the documentation is currently lacking?

akshatjoshi2001, 2022-11-04:
It would be great if the wiki could be completed with information about all modules just like the information available for branch predictors. For instance, while working on implementing my cache replacement policy on ChampSim I got confused on whether the full_addr argument in the update_replacement_state and find_victim functions includes the byte offset or not (that is whether full_addr is the block address or is it the address to a particular byte in the block). The name "full_addr" seems like a shortened version of full address which makes me feel that it should include the byte offset as well. But then I printed out the value of full_addr and compared it with the address attribute in the object of the BLOCK class, and it came out to be the same, which means that full_addr should be the block address instead. These ambiguities will not arise if a proper documentation covering all the major functions and their input parameters can be made.

ZJU-Andre, 2023-07-14:
Hi there, I'm using Champsim these days.
Firstly I didn't see the wiki page in this repository, and the paper was found when I looked for guide in google scholar. I'm wondering if it's posible to add wiki and paper link in readme? Thanks alot.

ngober, 2023-09-06:
I have redeployed the wiki to https://champsim.github.io/ChampSim/. It currently contains approximately the contents of the wiki. I would be happy to consider further pull requests to improve it, if anyone on this thread decides that that is the way they would like to contribute.

closed 2023-09-06 by ngober