ChampSim/ChampSim #17: E: Pin does not support instrumenting this conditional branch

zlukegord, 2019-08-08:
Hello again, I was able to get the code to compile by changing a couple things (string -> std::string, cerr -> std::cerr, etc). I was able to generate champsim_tracer.so, but now I am getting a new error when I try to run it. I'm not really sure what this error means or what I can do to fix it. Can anyone help?

My command line input:
../../pin2/pin2/pin -t obj-intel64/champsim_tracer.so -- ls

The error:
E: Pin does not support instrumenting this conditional branch: [ xbegin 0x7f4b2d69d8d8 ] with IARG_BRANCH_TAKEN

AbhishekUoR, 2020-08-19:
Did you get a way to solve this error?

GinoAC, 2020-10-22:
The provided ChampSim tracer currently supports Pin3.2, but from your command line input it looks like you're using Pin2.0. Try using Pin3.2 and see if the error still occurs.

closed 2020-11-05 by GinoAC