ChampSim/ChampSim #22: Core/Cache stalled on an STLB miss

vishalgupta97, 2020-01-17:
On an STLB miss, currently everything is stalled including the core, cache and DRAM. Why it is simulated like that? Wouldn't it be better if only that particular request is stalled and other request can proceed further?

sethpugsley, 2020-01-24:
I agree that it would indeed be better if only that particular request were stalled.  I think it was implemented this way as a shortcut so that page faults could cause some performance loss without having to actually implementing page tables and page walking.

I personally think it would be better to just get rid of this version of a page fault penalty, and err on the side of higher IPC rather than what's currently being done.  I just haven't gotten around to implementing this yet.

ngober, 2021-05-24:
Closing this issue since a page table walk has been added to the development branch.

closed 2021-05-24 by ngober