ChampSim/ChampSim #24: Instr being executed during Branch miss-prediction and ROB buffer flush

madnan92, 2020-01-28:
I just want to confirm that either the trace files includes the miss-predicted branch paths or not?
As from the code it looks like miss-predicted branch path is not included in the trace file and on branch miss-prediction only the branch miss penalty has been added and not the miss predicted instructions being implemented as ROB is not flushed in any way.
Can someone please confirm?

sethpugsley, 2020-01-28:
The trace files do NOT include any wrong-path information, because they were originally collected using a PIN tool, which cannot give information about wrong-path execution.  Therefore, the only side-effect of a branch mispredict is the mispredict penalty, and not the execution of any wrong-path instructions.  Because we aren't speculatively executing any wrong path instructions, we also never need to clear out anything from the ROB when there's a mispredict.  All we do is stall fetch (along the true path) until the mispredicted branch is executed.

madnan92, 2020-01-28:
This is exactly what I got from the code.
Thanks for confirmation.

closed 2020-01-28 by madnan92