ChampSim/ChampSim #30: RAW detection bug

chiragsakhuja, 2020-04-15:
I believe the RAW dependency check (reproduced below from O3_CPU::reg_RAW_dependecy) misses some dependencies. In particular, PIN uses different register IDs for equivalent registers of different sizes (e.g. RAX uses ID 10 and EAX uses ID 56 in PIN 3.13), so the check will not indicate there is any dependency if a binary uses mixed register sizes.

`if (ROB.entry[prior].destination_registers[i] == ROB.entry[current].source_registers[source_index])`

chiragsakhuja, 2020-04-16:
I think the quickest fix for this is actually in the tracer. In the `RegRead` and `RegWrite` functions, the lines that convert the `i` argument to `REG` type can be replaced with:
`
REG r = REG_FullRegName((REG)i);
`

A solution in the tracer itself would also make it agnostic of the version of PIN used to generate the traces. On the downside, existing traces would need to be regenerated, which might not make it worthwhile.

ngober, 2020-04-16:
Regenerating all of the traces is probably a cost we don't want to pay. I think we can get around it by creating a `arch_register` class that has an `operator==()` function, then converting the `destination_registers` and `source_registers` arrays (L139, L141 in inc/instruction.h) from `uint8_t` arrays to 'arch_register' arrays. 

If we do it that way, the 'arch_register' class can be specialized for different ISAs. You mentioned how RAX == EAX in x86, but also X0 == W0 in ARMv8, so we would need different specializations of the class. We can use templates for this, but that may be too awkward, since it might require template parameters to `O3_CPU`.

ngober, 2020-04-17:
I found an easier way to solve the problem. I added a parameter to the `reg_RAW_dependency()` function that performs a generic comparison, then defaulted it to an equality check. It would be simple to add a more robust check. Create a functor that does a harder comparison, like (using the example you describe):

    struct x86_reg_comp
    {
        bool operator() (const uint8_t lhs, const uint8_t rhs) const
        {
           if (lhs == rhs) return true;

            if (lhs == 10 || lhs == 56)
                if (rhs == 10 || rhs == 56)
                    return true;

            return false;
        }
    };

Then, L966, L973, L979 can be changed to `reg_RAW_dependency(i, rob_index, j, x86_reg_comp());`. I've tested both with some traces from IPC and it seems to work just fine.

chiragsakhuja, 2020-05-01:
Thanks for the quick response and solution! This solution looks safe and relatively non-intrusive. I made a simple PinTool to compare for which registers this problem arises, and it's a lot more than just EAX/RAX. Here's a LUT I generated using PIN 3.13. The index into the LUT is the register ID, and the output is the full register it maps to. For example, RAX has ID 10, so `reg_mapper[10] == 10`. Similarly, EAX has ID 56, so `reg_mapper[56] == 10` as well.

```
static constexpr uint8_t reg_mapper[] = {
    0, 0, 0, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15,
    16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 10, 10, 10, 9, 9,
    9, 8, 8, 8, 7, 7, 7, 5, 4, 3, 6, 25, 26, 3, 3, 4,
    4, 5, 5, 6, 6, 7, 8, 9, 10, 25, 26, 11, 11, 11, 12, 12,
    12, 13, 13, 13, 14, 14, 14, 15, 15, 15, 16, 16, 16, 17, 17, 17,
    18, 18, 18, 83, 84, 85, 86, 87, 88, 89, 90, 123, 124, 125, 126, 127,
    128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143,
    144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 123, 124, 125, 126, 127,
    128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143,
    144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159,
    160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175,
    176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191,
    192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207,
    208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219, 220, 221, 222, 223,
    224, 225, 226, 227, 228, 229, 230, 231, 232, 233, 234, 235, 236, 237, 238, 239,
    240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255
};
```

I'm not sure if the register mapping is dependent on the PIN version, but if so the LUT should be generated using PIN 3.2, which is recommended in the ChampSim README. I wasn't able to find a download for PIN 3.2, so I just used the latest version. If you want to reproduce the experiment, I can provide you with the PinTool code.

ngober, 2020-05-02:
I added your register mapping to my pull request. I ran a quick sanity check on perlbench_s-210B for 500M/1B instructions with no prefetchers. Adding in this register mapping drops the IPC from 1.011 to 0.98, which suggests to me that it's finding more RAW dependencies. Check out my pull request and see what you think.

closed 2022-03-09 by GinoAC