ChampSim/ChampSim #31: Choosing a build system

ngober, 2020-04-16:
I think I'm not alone in thinking that the current method of building ChampSim is clunky. Over the past year, there have been three pull requests opened to address issues (#10, #11, #27). The three of them are incompatible, but they all attempt to address the same issues: The current build system
- Does not allow for models that span multiple files. This leads to monolithic, non-reusable code.
- Does not permit compilation flags or parameters (such as CXXFLAGS) to be passed to the compiler.
- Requires that models be named `*.l2c_pref`, or similar. This causes problems with tools that would expect `*.cc` or `*.cpp`.
- Does not scale well with the number of options available. With the recent addition of L1I prefetchers, `build_champsim.sh` now takes 7 parameters, while it is likely that only one is important on any build.
- Rebuilds the entire project on every build. Why do we need to do this when we have tools like Make available to us?

I think that continuing to use Make is a smart idea. It is a standard and it is well-understood by developers. There are multiple ways to prepare the project for Make:
- Autoconf is the GNU standard for building projects, but is targeted more to enable compatibility, not to select between source files
- CMake is a possible alternative, but introduces a new dependency (#11)
- SCons is used by GEM5, and seems to be a powerful tool.
- A home-cooked solution may also be good, since it varies only marginally from what we have now. (#10 uses a Python script to parse a configuration file, #27 uses environment variables)

What features do we need out of our build system? What can we do to make it better?

itisntalex, 2020-06-11:
Nice to see that this question has been put on the table. I believe I would be nice to discuss it.

**Pros:**
 - As I'm the author of #11, I will advocate for something that relies on CMake. I believe it allows for much more flexibility than what is available with the current build system. It's relatively easy to extend it, by _extend_ I mean adding new replacement policies, prefetchers, etc. It is as simple as adding a new directory and `.cmake` file in it provide the files to use.
 - Going with this option also allows one to not rebuild every single source file every time, which is in my opinion much more efficient that what is currently available.
 - It would allow one to use pretty much any build system desired: Make, Ninja, etc. Assuming that it's going to be compiled for an Unix-based system.
 - Comes with a pretty simple and to the point GUI tools for anti-terminal folks around.

**Cons:**
 - On the other hand, I have to admit that CMake and especially its syntax is not straight-forward. It takes time to understand it and have mastery.
 - It introduces an additional dependency. Yes. But I will reply by the large adoption of this tool in big projects. CLion is shipped with it and uses it for its users' projects.

I believe CMake is a good option to go for and will be happy to share my build system if you guys are interested in it.

One last point, and that's a feature I like: wouldn't be amazing to have your knobs in a config file (JSON for instance) and provide it to the simulator instead of rebuilding many times with different knobs? That error-prone and pretty annoying. I'm working on such thing. Same, if you guys are interested. 😉

ngober, 2020-06-25:
I'm working on a different change which is going to depend on this issue being resolved, so I would like to move on this in the next couple of days. I had been waiting a bit because @JamesPC44 had told me he was working on a candidate build system that he wanted to put forward, but it looks like his solution was a lot like #10, just with some different ideas for config.sh.

I think I prefer #10 over #27 because it will extend more easily as the number of features grows. I don't like depending too heavily on environment variables because they may be invisible to the user.

I think I also prefer #10 over #11 because, while CMake may make a good solution for when the project grows in size, I don't think we are currently large enough to introduce a heavyweight solution.

sethpugsley, 2020-06-25:
The primary goal for ChampSim is to be simple for the beginner.  Please keep that in mind in all of these discussions.  I believe keeping it "simple" is more important than it becoming "good."

I'm not saying that to scare anyone away from trying to improve it.  I really appreciate all the effort lately people have put forward, but any change should make ChampSim better at its primary goal of simplicity.

If you propose a change that would make a grad student in a computer architecture class with no experience using CPU simulators take more than 10 minutes to get up and running with ChampSim, then I believe it's a bad change.  If you propose a change that would keep a total novice from becoming a ChampSim expert in under 2 months, then I believe it's a bad change.

I agree that it's dumb that it has to recompile everything after every small change, and that you need to keep around a separate binary for each configuration you're studying.  Any solution that improves these situations should maintain, or improve, its ease-of-use for people with zero experience.

There are a million and one ways that ChampSim could be changed to become a much more effective tool for an expert to wield in computer architecture research.  I want to make sure it's always accessible to students and others who need to go from 0-to-expert in the span of time between a microarchitecture championship call for papers and the submission deadline.

GinoAC, 2020-10-19:
The current build system has been updated to use the JSON-based configuration from pull request #69.

PR #69 (2020-10-19): JSON-based configuration and build system
Yet another build system. Resolves #31.

This one takes inspiration from all of the others #10, #11, #27 to produce one that I'm very happy with, due to its extensibility.

The config.sh script generates inc/champsim_constants.h, which contains all of the machine parameters previously scattered across various headers, src/core_inst.cc, which contains instantiations of the elements of the machine we're simulating (again, previously scattered across various source files), and a makefile.

The input to the configuration file is a JSON object and I have in [... cut]
closed 2020-10-19 by GinoAC