ChampSim/ChampSim #39: how to collect the  cloudsuite  traces?

huster-ckc, 2020-06-05:
how to collect the  cloudsuite  traces?
thaks

GinoAC, 2020-11-05:
Are you trying to download the already available cloudsuite traces or generate new cloudsuite traces using Pin?

huster-ckc, 2020-11-06:
> Are you trying to download the already available cloudsuite traces or generate new cloudsuite traces using Pin?

I want generate new cloudsuite traces using Pin .I think it's different from spec using pin.I hope you can give me some suggestions. Thank you for your reply

ngober, 2021-05-24:
The only difference between the cloudsuite traces and the current trace format is the inclusion of an `uint8_t asid[2]` field. If you want to generate traces that use this format, you would only need to modify the tracer to fill in this field. However, I'll warn that support for the cloudsuite traces has atrophied somewhat since they were used. Reviving support for ASIDs is on the development roadmap, but it probably will not happen soon.

closed 2021-05-24 by ngober