ChampSim/ChampSim #40: Reshaping ChampSim

itisntalex, 2020-06-11:
As I see that there has been quite a lot of issues about the structure of ChampSim, I was willing to make an addition to #31 #11 #10.

Here, I'm going to share with you the way I believe ChampSim should be structured.

As a matter of fact ChampSim, to my knowledge, is mostly used to try out replacement policies, prefetching mechanisms, and so on. It appears to me that it's quite unlikely during your research process that you'll change ChampSim internals. Then this part of the simulation infrastructure would better be fixed while changing components such as replacement policies, prefetching mechanisms, etc. would gain to be in distinct compilation units.

A third part that I would like to consider is the knobs or the many macros in the code to configure caches, CPU, and so on. Going on an exploration with them can rapidly end up being a compilation nightmare.

To wrap up, here is three key concepts that I would like to push forward to you guys:

1. **The main simulator infrastructure:** this part would be the actual ChampSim executable and it would be in charge of the initialization of the memory infrastructure, reading traces and more;
2. **Modules:** these would be the replacement policies, prefetching mechanisms, or any crazy thing you want. I reckon they should libraries loaded at runtime, containing the minimum code required to have its mechanism implemented and working;
3. **Configuration file(s):** this part would be an input telling the simulator which knobs to use when allocating structures, how the CPU(s) should be configured, which policies to use in which cache, etc.

**N.B.:** I understand that it might sound like a lot of work (it is actually), but I feel like it addresses some of the issues pointed out by @ngober in #10. However, to me, that sounds like a very good and ambitious milestone for ChampSim, making it more user-friendly and approachable.

ngober, 2020-06-23:
I have been thinking about this idea, and I like a lot of the ideas you've put forward. I would resist the idea that it is unlikely you will change the ChampSim internals, we have seen researchers change the internals to fit their ideas. But, I do like the idea of modules. How do you suggest we accomplish that?

GinoAC, 2020-10-19:
We've updated the current build system to use a configuration file format that this issue no longer applies to.

closed 2020-10-19 by GinoAC