ChampSim/ChampSim #42: ChampSim does not compile under Cygwin

ngober, 2020-06-16:
Alternate title: Windows users are people, too.

ChampSim does not compile under Cygwin due to the lack of support for POSIX threads. Currently, the only place pthreads are used is in src/main.cc, where it is used to spin up gzip in parallel and read the traces. If it does not take too much effort, it could be worth it to provide a Windows compatible solution in this one place. If we do, then ChampSim should run on Windows, too.

cpepi001, 2022-09-10:
I was able to compile ChampSim (5f8524e498bc1e4776c1c3fb7954a054bc5bcfc9) with Cygwin current version (3.3.6-1) with the following package versions

- make 4.3-1
- gcc-core 11.3.0-1
- gcc-g++ 11.3.0-1

One thing I had to change was `-std=c++17` to `-std=gnu++17` in the `config.sh:L688` file. However, I ran some benchmarks and the results deviate a bit comparing with a Linux version.

ngober, 2022-09-12:
The `-std=gnu++17` surprises me. I'm only aware of one instance of a GNU extension being used (in `tracereader.cc`), but it is hidden behind a preprocessor guard, so shouldn't prohibit `-std=c++17`. What is the compiler error, exactly?

The results deviating is something I've seen small indications about, just from things people tell me. It points to some instance of undefined behavior being reached in the program. I'm working on tracking those down, but what remains has been pretty elusive to get around my checks.

cpepi001, 2022-09-12:
This is the error I get when I execute `make`

```
g++ -Wall -O3 -std=c++17 -Iinc -MMD -MP  -c -o src/main.o src/main.cc
src/main.cc: In function ‘int main(int, char**)’:
src/main.cc:309:20: error: aggregate ‘main(int, char**)::sigaction sigIntHandler’ has incomplete type and cannot be defined
  309 |   struct sigaction sigIntHandler;
      |                    ^~~~~~~~~~~~~
src/main.cc:311:3: error: ‘sigemptyset’ was not declared in this scope
  311 |   sigemptyset(&sigIntHandler.sa_mask);
      |   ^~~~~~~~~~~
src/main.cc:313:41: error: invalid use of incomplete type ‘struct main(int, char**)::sigaction’
  313 |   sigaction(SIGINT, &sigIntHandler, NULL);
      |                                         ^
src/main.cc:309:10: note: forward declaration of ‘struct main(int, char**)::sigaction’
  309 |   struct sigaction sigIntHandler;
      |          ^~~~~~~~~
make: *** [<builtin>: src/main.o] Error 1
```

ngober, 2022-09-13:
Ah, I see. looks like Cygwin is not POSIX compliant. I did some tinkering and I realized that our signal handler... actually doesn't do anything at all. So it might be worth it to delete it. Try deleting the signal handling at the very beginning of main(), as well as the include of `<signal.h>`. If that compiles, submit it as a pull request and we can resolve this one.

ngober, 2023-06-01:
I think #323 may have removed the last vestige of GNU libraries in favor of more portable ones. Could you try again and see if the issue is resolved?