ChampSim/ChampSim #43: Latest version PIN Tools

dixit-kumar, 2020-06-16:
Please update link of pin tool to it's latest version.
Latest link: http://software.intel.com/sites/landingpage/pintool/downloads/pin-3.13-98189-gcc-linux.tar.gz

itisntalex, 2020-06-16:
Well, could you at least explain (or even better, justify) why the lastest version of ChampSim is required? Is there some new features? Bug fixes? etc.?

The link is dead by the way...

dixit-kumar, 2020-06-16:
Hey Alexandre,

First, sorry for the dead-link. Also, I am not aware about any new feature/bug-fixes in updated version of **PIN**.  I was just saying it as a minor suggestion for updation in readme. 

In case if there is something new in updated version of PIN. I will let you know. 
Here, updated link: https://software.intel.com/sites/landingpage/pintool/downloads/pin-3.13-98189-g60a6ef199-gcc-linux.tar.gz

By the way, ChampSim-Tracer PIN-TOOL is perfectly fine. It doesn't require any updation.

itisntalex, 2020-06-16:
So, I had a look myself about any bug-fixes or new features in the latest release of PIN and so far they didn't update the release notes. Kind of funny I would say 😄 .

I believe it could be nice to have it updated. However, I guess most people around here, just use any version available able to do the job. At least I do... 

But if you really want it to be updated, just drop a pull-request to the `develop` branch and project maintainers will let you know. 😉

closed 2020-06-17 by dixit-kumar