ChampSim/ChampSim #47: RAW dependencies are broken

lpha-z, 2020-06-22:
I think following code (in the `develop` branch) breaks RAW dependencies.
https://github.com/ChampSim/ChampSim/blob/5845533318dbb65754034036b5763acb96bbf9d2/src/ooo_cpu.cc#L1112

I test `./bin/bimodal-no-no-no-no-lru-1core -warmup_instructions 1 -simulation_instructions 1000 -traces ipc1_public/client_001.champsimtrace.xz`, and got the result of following:

```
(omit)
[ROB] reg_dependency instr_id: 200 is_memory: 1 load  reg_index: 1
[ROB] reg_dependency instr_id: 200 is_memory: 1 store reg_index: 67
(omit)
[ROB] reg_dependency instr_id: 202 is_memory: 1 load  reg_index: 67
[ROB] reg_dependency instr_id: 202 is_memory: 1 store reg_index: 67
(omit)
[RTL0] operate_lsq instr_id: 200 rob_index: 200 is popped to RTL0 head: 26 tail: 28
[RTL0] operate_lsq instr_id: 202 rob_index: 202 is popped to RTL0 head: 27 tail: 28
(omit)
```

The instruction with id200 is a load instruction and it reads from reg1 and writes to reg67.
The instruction with id202 is a load instruction and it reads from reg67 and writes to reg67.
The instruction with id202 is woken up at the same time with the instruction with id200, but this violates RAW dependency (writing to reg67 will be completed after a few cycles).

I didn't know what the above code meant, so I didn't know how to fix it.
(I estimated it's intended to handle idioms like `xor eax, eax`. The problem is came from not examining memory operands. Even if the problem is solved, it is impossible to distinguish `add eax, 1` from `xor eax, eax` in the current trace format.)

sethpugsley, 2020-06-22:
First of all, you are right that the current ChampSim trace format makes many common code patterns indistinguishable from each other, even though they may have different dependencies between register and load address generation.

I recently added the bit of code you cited above in an attempt to differentiate between these identical-looking situations.  Ultimately, it's just a guess, but I was looking for ways to get around ChampSim's trace format's penalty where it always assumes the worst, i.e., that all load address generation is dependent on ALL register reads in an instruction.  The way ChampSim worked before was definitely wrong.  This is also definitely wrong, but it improves IPC a tiny bit by sometimes allowing load address generation to not be dependent on all registers in the instruction.

What's unclear from your example is what you mean by the instructions "waking up."  The intent of my code is that instruction 202's load can initiate before instruction 200 writes to register 67 and clears the dependence between them.  What I DON'T intend is for instruction 202 to actually "execute" before (or on the same cycle as) instruction 200.  Is that what you're observing, or is it just that instruction 200 and 202 are able to issue their loads on the same cycle?  Just to reiterate, I'm fine with the loads both issuing on the same cycle, but I'm not fine with both instructions "executing" on the same cycle.  Which of these situations is occurring?

I appreciate all your work to track down these kinds of bugs.  Keep up the good work!

lpha-z, 2020-06-22:
Sorry for unclear wording. "Waking up" is intended as `ROB.entry[i].reg_ready = 1;` at L1813.
https://github.com/ChampSim/ChampSim/blob/5845533318dbb65754034036b5763acb96bbf9d2/src/ooo_cpu.cc#L1813

200's load and 202's load are executed on the same cycle.

```
[L1D_RQ] add_rq instr_id: 200 address: 3f7f4ea3174 full_addr: fdfd3a8c5d34 type: 0 head: 23 tail: 24 occupancy: 1 event: 2402 current: 2397
[L1D_RQ] add_rq instr_id: 202 address: 3f7f4ea3177 full_addr: fdfd3a8c5dcc type: 0 head: 23 tail: 25 occupancy: 2 event: 2402 current: 2397
```

As another clearer example, 701's load instruction is executed before 696's ALU instruction executed.

```
[ROB] reg_dependency instr_id: 696 is_memory: 0 load  reg_index: 19
[ROB] reg_dependency instr_id: 696 is_memory: 0 load  reg_index: 4
[ROB] reg_dependency instr_id: 696 is_memory: 0 store reg_index: 67
[ROB] reg_dependency instr_id: 697 is_memory: 0 load  reg_index: 26
[ROB] reg_dependency instr_id: 697 is_memory: 0 load  reg_index: 25
[ROB] reg_dependency instr_id: 697 is_memory: 0 store reg_index: 26
[ROB] reg_dependency instr_id: 698 is_memory: 1 load  reg_index: 1
[ROB] reg_dependency instr_id: 698 is_memory: 1 load  reg_index: 18
[ROB] reg_dependency instr_id: 698 is_memory: 1 store reg_index: 1
[ROB] reg_dependency instr_id: 699 is_memory: 0 load  reg_index: 2
[ROB] reg_dependency instr_id: 699 is_memory: 0 store reg_index: 2
[ROB] reg_dependency instr_id: 700 is_memory: 0 load  reg_index: 26
[ROB] reg_dependency instr_id: 700 is_memory: 0 load  reg_index: 25
[ROB] reg_dependency instr_id: 700 is_memory: 0 store reg_index: 26
[ROB] reg_dependency instr_id: 701 is_memory: 1 load  reg_index: 67
[ROB] reg_dependency instr_id: 701 is_memory: 1 store reg_index: 67

[LQ] add_load_queue instr_id: 701 is added in the LQ address: c1781dd8 translated: 0 fetched: 0 index: 1 occupancy: 3 cycle: 7098
[L1D_RQ] add_rq instr_id: 701 address: 3f7f4ea3177 full_addr: fdfd3a8c5dd8 type: 0 head: 45 tail: 46 occupancy: 1 event: 7105 current: 7100
[L1D] check_hit instr_id: 701 type: 0 addr: 3f7f4ea3177 full_addr: fdfd3a8c5dd8 tag: 3f7f4ea3177 data: 0 set: 55 way: 0 lru: 0 event: 7105 cycle: 7105
[ROB] complete_execution instr_id: 701 is_memory: 1 branch_mispredicted: 0 fetch_stall: 0 event: 7105 current: 7105

[ROB] do_execution non-memory instr_id: 696 event_cycle: 7209
[ROB] reg_RAW_release instr_id: 696 releases instr_id: 701 reg_index: 67 num_reg_dependent: 0 cycle: 7209
[ROB] complete_execution instr_id: 696 branch_mispredicted: 0 fetch_stall: 0 event: 7209
[ROB] do_memory_scheduling instr_id: 701 rob_index: 349 scheduled all num_mem_ops: 0
```

701's load is dependent on 696's ALU instruction, which writes to reg67.
696's ALU instruction is executed at 7209th cycle.
701's load is executed at 7100th cycle and completed at 7105th cycle.
This is a dependency inversion.

sethpugsley, 2020-06-22:
I went through the code and figured out what's going on, and I have some idea of what it would take to fix this problem, but I don't think it's worth it.

I think it would be better to abandon my plan to try to identify some niche situations where loads can go ahead of inter-instruction register dependencies being resolved, and just go back to how it used to be.  I propose that you replace the mem_reg_dependence_resolved() function with this:

uint8_t O3_CPU::mem_reg_dependence_resolved(uint32_t rob_index)
{
  return ROB.entry[rob_index].reg_ready;
}

Would you please test out if that fixes the problem you're seeing, and report back on the performance impact?  If this seems like a good solution to you, would you please implement this change and push it to the develop branch?  Thanks.

lpha-z, 2020-06-23:
I tested 2 methods on the 50 A64 ipc1 public traces and the 46 x86 SPEC2017 simpoints that was used in dpc3.

The simulation model is `hashed_perceptron-next_line-next_line-spp_dev-no-lru-1core`.

|methods|dep. estimation|geomean of 50 A64 traces|geomean of 46 x86 traces|
|:-|:-|:-|:-|
|`develop`|optimistic|IPC=1.141|IPC=0.957|
|Seth's comment|pessimistic|IPC=1.051|IPC=0.949|

The optimistic version improves performance in

* almost all of A64 traces (up to 55%, server009), and
* x86 traces of perl, gcc, mcf, and omnet (up to14%, gcc1850).

This improvement is a false improvement, and it should be fixed.

I have two impressions:

* In A64 traces, RAW dependencies are completely broken.
* In x86 traces, I found few dependency inversion (but I found in omnet141 trace, 172's load is executed before 169's load). However, there is another problem. X86's `pop` sequences can be pipelined, but they are executed completely sequential in current two implementations, which has a significant performance impact. I think it's worth to treat this case specially.

I'm currently thinking of a way to handle `pop` sequences well. Is this useful for ChampSim?

lpha-z, 2020-06-23:
I understand your intention now. The problem is came from the difference between A64 and x86. In x86, "writes to reg1, read from reg1, reg2, and memory" instruction may be a `add rax, [rbp+8]`-like instruction, and its address generation does not depend on reg1. In contrast, in A64, which is a RISC ISA, "writes to reg1, reads from reg1 and memory" must be load instruction and its address generation must depend on reg1.

sethpugsley, 2020-06-23:
I hadn't considered the differences between traces that came from A64 and x86_64 when I originally introduced this change.  With that in mind, I think it's very important that we remove the optimization I made that introduced this error.

If there is another way to handle pop sequences without modifying the trace format, then I would be very interested in that.  It may be difficult to identify pop instructions right now, because apparently not all of our traces use the same stack register, IP, and flags register numbers, even though they are hard-coded to fixed values*.  We need to identify at run-time the register number of our stack pointer (along with IP, and flags), and then we may be able to do some special optimizations for pop sequences.

*these hard-coded values are correct for all of the IPC-1 traces, and most of the DPC3 traces, but there are a few DPC3 traces where they are wrong.

ngober, 2020-06-23:
We could introduce a command-line flag that tells the core which ISA it's running. For example, `-m x86_64` would set the stack pointer index, flag register index, etc to x86 values, while `-m A64` would set the same for A64. This may also play interestingly with #32. We could introduce multiple versions of this patch, one for each ISA we support.

In the long term, if we ever change the trace format, this would be good information to include in a file header.

lpha-z, 2020-06-23:
Command-line-flag-based solution looks good. I will push the pessimistic version for now, but I will continue to seek ways to handle the `pop` sequence.

By the way, there is yet another problem. Unlike x86 traces, A64 traces' branch instructions don't seem to depend on the flags register. Therefore, branch misprediction is resolved immediately. This leads to overestimation of performance.
I want to solve this problem too, but I have no good idea.

ngober, 2020-06-23:
I'm willing to bet that it isn't true that they don't depend at all on the flags register; A64 has the B.GE (and similar) instructions that branch based on a flags register, but it also has the CBZ instruction that checks for zero in an arbitrary register. It is likely that we simply have the flags register incorrect for A64. It may be that (and I am probably straying off topic for this thread) an indirect branch and a conditional branch are indistinguishable in our current setup.

sethpugsley, 2020-06-23:
We are straying off topic, but indirect and conditional branches are distinguishable in the A64-based IPC-1 traces.  Look at the branch statistics at the end of a run to see how many of each type were encountered.

lpha-z, 2020-06-23:
I run `xz -d ipc1_public/client_001.champsimtrace.xz -c | hexdump -v -e '8/8 "%016x "' -e '"\n"' | head` and I see the 3rd instruction is an instruction that reads from reg25 (REG_FLAGS) and reg26 (REG_INSTRUCTION_POINTER), writes to reg26 (REG_INSTRUCTION_POINTER), reg1, and reg1.

* This instruction is a branch instruction because the next PC is not PC+4. 👌 
* This instruction uses reg26 as instruction pointer though it is an A64 instruction. 🤯 
* This instruction seems to use reg25 as condition but no instruction seems to write to reg25. This is the root of the problem, and in a sense, it looks like the dependencies aren't followed.

ngober, 2021-02-26:
I think this was merged in #54. Closing.

closed 2021-02-26 by ngober