ChampSim/ChampSim #49: A new approach to the virtual memory system

sethpugsley, 2020-06-24:
ChampSim's approach to a virtual memory system is currently very basic.  It provides the basic functionality of separating the address spaces of multiple processes, but it doesn't correctly model the performance impact of offering this support.

ChampSim's current virtual memory system consists of 4 parts: an L1 ITLB, and L1 DTLB, an STLB, and a va_to_pa() function.  The L1 TLBs plug into the STLB, and when there's an STLB miss, the va_to_pa() function is invoked to get the correct translation, which is then cached in the appropriate TLBs.

The va_to_pa() function has always been implemented in a "magic" sort of way.  It used to impose a magic, draconian penalty whenever it was invoked that would freeze all functions of the core, even those that had nothing to do with virtual memory.  I recently bypassed that to eliminate the penalty altogether, because I believe an unrealistically small penalty was still more realistic than the extreme penalty that was in place.

So what's missing in ChampSim's virtual memory system?  Page walkers.  There was a recent pull request to introduce page walkers, but these page walkers also worked in a very magical way.  They would instantaneously update caches, and worked with hard-coded fixed latencies.

Real page walkers should generate memory transactions that traverse the page table, naturally caching everything they touch in the existing cache hierarchy.  They should be state machines that work over many cycles to return an answer that's added to the TLBs.

Here is a very rough outline of what I propose:

*** a virtual memory class ***

At initialization time, you specify the size of the physical address space, the number of processes that will be sharing the physical memory, and the number of page table layers.  There will of course be a function similar to the existing va_to_pa() function that can magically give you a translation, but also within the physical address space will exist the page tables for all of the processes, and this kind of capability:

get_pte_physical_address(process_number, virtual_address, layer_number)

This is what the page walker will use to traverse the page table.  First it will start at get_pte_physical_address(PID, VA, 0), which will return the root of the page table, and then when that returns it will move on to fetch get_pte_physical_address(PID, VA, 1), and so on.  When the leaf node of the page table returns to the page walker, then the va_to_pa() function can be used to complete the translation.

*** TLBs ***

TLBs are currently the same type as caches, and work in exactly the same way.  I propose making them a sub class that inherits from caches instead, with these changes:

1) TLBs can be backed up by another TLB, or by a page walker.  You can't have a TLB that just hangs out by itself.
2) TLBs can be attached to any cache.  The project I'm working on now wants to add virtual address prefetching at all levels of the hierarchy, so I want to add TLBs at the L1, L2, and L3 caches.  Each of these TLBs would be backed up by its own page walker in this model.  Or maybe it would be possible to have the L2 and the L3 plug into the L1's STLB.  Who knows?  The point is, I want TLBs and address translation capability to exist at more than a single point in the hierarchy.

*** page walkers ***

Page walkers would be finite state machines that traverse the page table of their core's process.  The page walker would need to be forwarded the addresses of all cache fills and cache hits for the level of the hierarchy where they live, and as soon as one page table entry shows up at the cache, then it can generate a new memory transaction to read the next level page table entry.

Page walks with a lot of cache misses need to take longer than page walks with cache hits, and page walks need to generate real cache and DRAM traffic that consumes bandwidth and impacts the latency of other memory requests.

---

I'm sure I've left out millions of necessary details, but I just wanted to get a conversation started.  In summary, I want to add 3 more classes to ChampSim to support virtual memory: a virtual memory class, TLBs (which would inherit almost all of their functionality from caches), and page walkers.

Does this plan sound reasonable?  What's missing?

ngober, 2020-06-24:
I have been thinking about this today, and I think I have something that feels like a solution which requires only minimal changes.

Currently, each cache keeps a pointer to its lower level, its upper instruction level, and its upper data level. We could add to it a pointer to its "address resolver". When a cache needs to perform an address translation, it adds a packet into its translator's RQ. The address resolution does not complete until data is returned from the address resolver. This fits your model of "TLBs can be attacked to any cache".

The page walker is of course the hardest part, but I see it as a sort of adapter. It takes requests from a TLB and turns them into a sequence of requests to (an arbitrary level of) the main cache. Currently, this last part is the false thing, but "object that takes requests from a cache-like object and turns them into requests to another cache-like object" sounds very much like what a cache does. I suppose it wouldn't be too hard to base the interface off of the `MEMORY` class.

This idea requires returning to change slightly. If the data is instruction, it goes to the upper I-cache; if it is data, it goes to the upper D-cache; if it is translation, it goes to the TLB. Note that, as is currently true in the LLC, the upper-I, upper-D, and upper-TLB might not be three different caches.

Here is the model of how the system is currently laid out, with my addition of a page walker.

![A model of ChampSim's cache hierarchy. The core connects via to each of the L1I, L1D, ITLB, and DTLB. Each of these connects to their second-level cache. The L2C connects to the LLC, then the DRAM, while the STLB connects to the page walker, and possibly then the L2C.](https://user-images.githubusercontent.com/4713238/85621084-56d67b80-b62a-11ea-9372-39dcd06e9fea.png)

I included a dotted line coming out of the page walker because I'm not sure where it would send its requests, but that would be a trivial change. It is pretty easy to see how any level of the cache could connect to any particular TLB, and the TLBs can still have a request pathway to the DRAM. Under this model, I don't think a separate "Virtual Memory" class is necessary. It can simply be handled by DRAM.

biswabandan, 2020-06-28:
This sounds great. My two cents (if possible): 
(i) Multiple PTWs instead of just one. 
(ii) Since ChampSim's L1 is Writeback, PTWs should go through L1 cache too.

ngober, 2020-06-28:
Should PTWs go through the L1I or L1D? Or both, depending on whether the request is for instructions or data?

gvavou5, 2020-06-29:
PTWs always go through the L1D cache since page table entries are treated as data by the processor [Computer Architecture: A Quantitative Approach - Appendix L].

ngober, 2020-07-08:
I have been working on a page walker, and I feel reasonably good about it at this point, but it still needs some work. 

I have run into a bit of a stopping block and I want to check before I make a big change. In general, prefetchers are attached to the cache and operate in the physical space (although with the recent merge of #52, we now have the option for either physical or virtual). However, the L1I prefetcher is different. It lives in the out-of order core and is driven by the frontend (this has been true since d0e59b64f00494ab800d43dfb388603d3627fc3c). It operates entirely in the virtual space.

As part of implementing the page walk, I am eliminating the `va_to_pa()` function. All translations must go through the TLBs, which I think is correct. If this is the way we want to do it, then the L1I prefetcher interface needs to change considerably to match the other cache prefetchers. Are we okay with this change?

ngober, 2021-03-29:
Resolved with #121.

closed 2021-03-29 by ngober