ChampSim/ChampSim #56: collected traces can not run on champsim

qianlong-zhang, 2020-07-30:
Hi, 
I collected traces using the following commd: 
pin-3.2-81205-gcc-linux/pin -t /home/zhangqianlong/MyPHD/ChampSim/tracer/obj-intel64/champsim_tracer.so -o ls .champsim -- ls
then :
tar cfJ ls.champsim.tar.xz ls.champsim

and run:
./bin/perceptron-large-no-bingo_16k-lru-4core  -warmup_instructions 10 -simulation_instructions 10000 -traces ls.champsim.tar.xz ls.champsim.tar.xz ls.champsim.tar.xz ls.champsim.tar.xz 

erros like this: 
CPU 0 runs ls.champsim.tar.xz
CPU 1 runs ls.champsim.tar.xz
CPU 2 runs ls.champsim.tar.xz
CPU 3 runs ls.champsim.tar.xz
[ROB_ERROR] add_to_rob ip is zero index: 1 instr_id: 1 ip: 0
perceptron-large-no-bingo_16k-lru-4core: src/ooo_cpu.cc:321: uint32_t O3_CPU::add_to_rob(ooo_model_instr*): Assertion `0' failed.
Aborted (core dumped)

what's the problem? thanks

GinoAC, 2020-11-12:
Hi, try compressing the trace into just a ".xz" format without tar-ing it. If the problem continues, make sure that any changes you've made maintain the trace format defined in the original tracer.

closed 2020-11-20 by GinoAC

DeeprajPandey, 2020-12-13:
Hi @qianlong-zhang, you are supposed to use `xz` instead of `tar` to generate the compressed trace files.

After this step
> pin-3.2-81205-gcc-linux/pin -t /home/zhangqianlong/MyPHD/ChampSim/tracer/obj-intel64/champsim_tracer.so -o ls .champsim -- ls

You should try
```sh
xz ls.champsim
```

And you will have a `ls.champsim.xz` file in the same directory which will work with ChampSim.

Note: In some kernels and pin versions, Pin throws an error like `Pin does not support instrumenting this conditional branch: [ xend ] with IARG_BRANCH_TAKEN`. If all you are doing is trying some random traces, try with a random program instead of using ls.