ChampSim/ChampSim #67: Getting the right number of L1D misses

mahmoodn, 2020-09-15:
Hi
As I look at the code, I see that read miss stat in [cache.cc](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L805) is incremented via 

MISS[RQ.entry[index].type]++;

By Simply filtering the stat for `L1D` as below

```
if (NAME == "L1D")
    cout << "M\n";
MISS[RQ.entry[index].type]++;
```
in the end of the simulation, I count the number of 'M' symbols and get 100. 

Then I expect that TOTAL stats in main.cc be the same as the number of 'M' symbols. However, it is very small compared to 100.

Tracking the code, I see that in the final stat phase, `cache->roi_miss[cpu][0]` is printed ([main.cc](https://github.com/ChampSim/ChampSim/blob/master/src/main.cc#L51)). That gets the value from `cache->roi_miss[cpu][i] = cache->sim_miss[cpu][i];` ([main.cc](https://github.com/ChampSim/ChampSim/blob/master/src/main.cc#L33)) and then we see value is incremented by `sim_miss[fill_cpu][MSHR.entry[mshr_index].type]++;` ([cache.cc](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L42)).

So, it seems that what is printed in the output of the run as L1D misses, is a stat for MSHR and not L1D. 
Any comment on that?

ngober, 2020-09-15:
The `sim_miss` counters are incremented when a line is filled, rather than when it is actually missed. So, especially if you are only running for a short number of instructions, you may find that many cache lines have missed (compusively), but their fills may be in-flight, so they are not counted in the final output.

I have suspected for some time that there is something strange going on with the `HIT/MISS/ACCESS` family of counters versus the `sim_hit/sim_miss/sim_access` family. If you would be willing to look into it, resolve any incorrect counts, and submit a pull request, it could be very beneficial to the project.

mahmoodn, 2020-09-16:
Good hint.
In fact, if you look at the the read hits, you will see that [sim_hit](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L590) and [HIT](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L623) are incremented. However, for read miss, only [MISS](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L805) is incremented.

For writes, it is fine and you will see [HIT](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L289), [sim_hit](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L253), [MISS](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L513) and [sim_miss](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L482).

So, the absence of sim_miss in handle_read() is a bug.
I added that and it now looks like
```
if (NAME == "L1D")
     cout << "M\n";
MISS[RQ.entry[index].type]++;
ACCESS[RQ.entry[index].type]++;
// COLLECT STATS
sim_miss[read_cpu][RQ.entry[index].type]++;
sim_access[read_cpu][RQ.entry[index].type]++;
```
In the output, I see
```
L1D TOTAL     ACCESS:    3479203  HIT:    3476375  MISS:       2828
L1D LOAD      ACCESS:    1654431  HIT:    1651707  MISS:       2724
L1D RFO       ACCESS:    1824772  HIT:    1824668  MISS:        104
L1D PREFETCH  ACCESS:          0  HIT:          0  MISS:          0
L1D WRITEBACK ACCESS:          0  HIT:          0  MISS:          0
```
and when I count M symbols, it is 2366. Still there are some differences.
The command I am using is `./run_champsim.sh bimodal-no-no-no-no-lru-1core 1 10 600.perlbench_s-210B.champsimtrace.xz`

mahmoodn, 2020-09-16:
UPDATE:
I see that the difference comes from some situations were `MISS` is not incremented while `sim_miss[read_cpu][RQ.entry[index].type]++;` is incremented. For example, in the beginning, they are incremented neck by neck. That is:
0 and 0
1 and 1
2 and 2
3 and 3
4 and 4
5 and 5
In the next round by setting a breakpoint at MISS statement, I see
5 and 6
which means some where in the code `sim_miss[read_cpu][RQ.entry[index].type]++;` has been incremented. Grepping `sim_miss` and `[read_cpu]` shows that line only in the source codes. That means, no where in the code, there is another instance for `sim_miss[read_cpu][RQ.entry[index].type]++;`.

They questions is why the counters are
5 and 6 
now?

Fortunately, I was able to catch that. In fact, I see that in the [handle_fill()](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L153), before executing `sim_miss[fill_cpu][MSHR.entry[mshr_index].type]++;`, the counters were
5 and 5
but after executing that line it becomes
5 and 6.

So, when it reaches handle_read(), MISS is 5 and sim_miss is 6.

Now, the question is these two statements seems to be different:
```
sim_miss[read_cpu][RQ.entry[index].type]++;
```
and 
```
sim_miss[fill_cpu][MSHR.entry[mshr_index].type]++;
```
But why incrementing the second affects the first one?
In fact, by watching the variables, for both of them, I see `sim_miss[0][0]`.
You may want to look at the [image], too.

mahmoodn, 2020-09-16:
So, I think the problem is that in handle_fill(), [sim_miss](https://github.com/ChampSim/ChampSim/blob/master/src/cache.cc#L153) is incremented, while there is no statement for MISS increment.

I guess there should be a WQ structure similar to RQ, but I don't see that. So, I am not sure if `MISS[MSHR.entry[mshr_index].type]++;` is correct in handle_fill().

Any comment?

mahmoodn, 2020-09-16:
@ngober 
I guess the way you mentioned how misses are counted needs some explanation.
You said that sim_miss right now counts the misses when a block is filled in cache. However, I think as soon as we see a read, misses in cache, the appropriate counter should increment. Don't you agree?

Assume, we are reaching the end of simulation. Here is what happens with current methodology:
1- A read, misses in cache.
2- A packet is sent and block will arrive later.
3- Simulation ends
4- Since there was no fill, miss is not incremented.

However, we have to note that a miss occurs before the end of simulation. In fact, if that accesses hit in cache, we were able to execute an instruction. So, that miss must be included in the counter.
Do you have any comment?

ngober, 2020-09-16:
I think that the way it currently works (that is, marking a miss on fill, rather than on miss) is more correct. When we are counting instructions, we count them as they are retired. So, if I run the simulation for 500 million instructions, I may issue more than that many instructions, but I will stop after 500 million are retired. I can then calculate stats such as misses per thousand based on how many of the 500 million instructions missed the cache. If we instead counted misses when we do tag matching, we may overstate the number of cache misses for a fixed number of instructions, especially at higher caches. Counting on fill seems to be the best strategy to me.

mahmoodn, 2020-09-16:
It that case, the sim_miss in handle_fill should be use (at it is now written at the end of the run). Therefore, the MISS counter in handle_read is misleading.
I guess these two counters were introduced at different stages of development, but one (MISS) has not been matured.

closed 2020-09-22 by mahmoodn