ChampSim/ChampSim #71: sim_miss is flushed in the middle of the run

mahmoodn, 2020-09-17:
I just added a cout after sim_miss in handle_fill() to print the value of the counter. Like this

```
sim_miss[fill_cpu][MSHR.entry[mshr_index].type]++;
cout  << current_core_cycle[fill_cpu] << " "
         << NAME << " "
         << fill_cpu << " "
         << unsigned(MSHR.entry[mshr_index].type) << " "
         << sim_miss[fill_cpu][MSHR.entry[mshr_index].type]
         << "\n";
```
In the output I see something like this:

```
388327 L1D 0 0 718
391859 L1D 0 0 719
393049 L1D 0 0 720
393158 L1D 0 0 721
398761 L1D 0 0 722
404647 L1D 0 0 1
413331 L1D 0 0 2
413545 L1D 0 0 3
462999 L1D 0 0 4
471030 L1D 0 0 5
479273 L1D 0 0 6
485725 L1D 0 0 7
```
Sounds weird... 
As you can see the cycle number is increasing which shows a progress in time. The NAME, fill_cpu and MSHR.entry[mshr_index].type are also constant. However, somewhere in the middle of the run the stat becomes zero.
I would like to know if you agree with me that this must not happen in the middle of simulation or there is a logic for flushing/resetting sim_miss in the middle of the run.

ngober, 2020-09-17:
I just grepped the codebase, and the only time that sim_miss is set to zero is at the end of warmup. Were you running 400,000 instructions of warmup, by any chance?

mahmoodn, 2020-09-17:
Right. I thought that the stats are counted after the warmup.
Thank you.

closed 2020-09-17 by mahmoodn