ChampSim/ChampSim #73: Instruction pointer in champsim

mahmoodn, 2020-09-22:
I would like to know if `MSHR.entry[mshr_index].ip` means the instruction address (PC) or not. Based on my understanding form the code, it should be the same as PC. For libquantum_964B, I have traced 100M instructions and for memory access and see that there are only 36 PCs for memory blocks. While it is related to the code base of libquantum, I want to be sure about that.

ngober, 2020-09-29:
Yes, the IP (instruction pointer) is synonymous with the PC (program counter). I agree that it is odd that there are only 36 unique PCs for memory blocks, but there could be any number of reasons for that. If you are looking at a data cache, for example, you will only see the PCs of instructions that access the memory, which may be far less than the number of instructions.

mahmoodn, 2020-09-29:
Thanks for the comment.

closed 2020-09-29 by mahmoodn