ChampSim/ChampSim #77: ChampSim build Failed : Line 100 and 101 Command Not found while executing build_ChampSim.sh

dwarakanadhas, 2020-10-26:


GinoAC, 2020-10-26:
This appears to mean that you do not have **make** installed on your machine. Try installing it and see if that resolves the issue.

dwarakanadhas, 2020-10-26:
Thanks for your reply. Can you please elaborate?

GinoAC, 2020-10-26:
Line 100/101 includes the make command, which runs the Makefile included with ChampSim. Since the make command is not found, it may mean you do not have make installed on the computer you're building ChampSim on and you will need to install it to successfully complete the build script.

dwarakanadhas, 2020-10-26:
Got it... Thank u so much for ur reply. I will install make and try to execute.

dwarakanadhas, 2020-10-26:
i have installed make file and tried to execute. it is giving the same error.

GinoAC, 2020-11-05:
Are you still having this issue? If so, what OS are you trying to compile on?

closed 2021-05-06 by GinoAC