ChampSim/ChampSim #83: ChampSim Prefetcher Address Space

manaal22152, 2020-11-10:
Does the prefetcher code in ChampSim operate in Virtual address space or physical address space? I mean  for the function:
void CACHE::l1d_prefetcher_operate(uint64_t addr, uint64_t ip, uint8_t cache_hit, uint8_t type)
 is the address addr virtual or physical?

ngober, 2020-11-10:
With the exception of the L1I, all prefetchers operate in the physical space. There is a feature on the development branch to enable virtual-address prefetching, but it is rudimentary and not complete yet.

manaal22152, 2020-11-11:
Is there a way to convert that physical address to virtual address at the prefetcher level?

ngober, 2020-11-11:
Not easily. You would have to do a lookup into the page table, which will be unrealistic. If you want information about the virtual addresses, even to simply get a physical-to-virtual mapping, you should use the development branch. Be wary, though, since it is our unstable branch and things are occasionally ~~breaking~~improving.

closed 2020-11-13 by ngober