ChampSim/ChampSim #84: Code Formatting

ngober, 2020-11-10:
There has been considerable talk on #38 about what code format style to adopt and how to implement it. I'm creating this issue to collect all of the discussion in a more topical place.

First, why ChampSim needs to care about code formatting: its purpose is to be an accessible tool for rapid innovations in computer architecture research, particularly in memory prefetching and replacement. In order to be accessible, it has to be readable and comprehensible. Therefore, any style that ChampSim adopts must prioritize comprehensibility.

Any style tool that ChampSim uses must also not be a burden to a novice programmer. We see many people using ChampSim that have never worked on a C++ project before. ChampSim is designed with modularity, that is, you can add your own custom prefetchers, branch predictors, and replacement policies. These modules, being primarily used in-house, need not respect any ChampSim format guidelines and should not be required to. Yet, contributors to the codebase should be required to maintain proper style (for self-evident reasons). Therefore, any style tool that Champsim uses must be as optional as possible for researchers while being as required as possible for maintainers.

In the other thread, two toolchains have been suggested:
 - [editorconfig](https://editorconfig.org/)
 - [ClangFormat](https://clang.llvm.org/docs/ClangFormat.html)

Let's use this thread to discuss two things: What style to adopt and what toolchains to use to implement it.

sethpugsley, 2020-11-10:
I played around with ClangFormat, and I liked how easy it was to use to format my local code files.  I don't know how easy it is to integrate with the project and enforce on other people.  I wasn't a big fan of any of the default styles, though.  They all seemed too dense and robotic.  As is evident by a lot of the code in ChampSim, my style is more ... let's say full of vertical white space.

If I had to pick any of them, I would choose llvm.  I know it's customizable, and we could create our own style for this project, but I also don't want to be a special snowflake who needs to have his own favorite style enforced on everyone else.  I think llvm is good enough.

One option is to not enforce a style on all future commits though an automated process, but to just have regular (say, monthly) re-formatting commits that don't contribute any new code, and just enforce a style format.  We would have to have some kind of regression test that would require the pre- and post-formatting code to produce identical output (we should probably have something like this anyway).  This would allow people to do whatever they want, and not have anyone meddling with what they can or cannot commit to a local repository.

GinoAC, 2020-11-11:
ClangFormat seems straightforward and it'd be nice to not have to force others to change their style to match a style-guide. I don't think it'd be difficult for new users to set up and it may even reinforce good style-practices for any new C++ coders.

I think it may be worthwhile to require new pull requests to have a final commit that is just a formatting pass. My fear is that a monthly formatting commit will inevitably cause a bug that requires stepping through all commits from the previous month to find the culprit. Even if this is easy to correct, iteratively updating the formatting at the end of a pull request review doesn't seem difficult and could prevent some headache later on.

ngober, 2020-11-13:
I tried ClangFormat, and I love how easy it is to get working, especially with how it uses the LLVM tokenizer. I would be fine adopting it as our tool of choice. I did some digging, and I found [this GitHub Action](https://github.com/DoozyX/clang-format-lint-action) that will apply ClangFormat automatically to any pull request. That's a pretty good setup, I think. Should I try to get that working?

sethpugsley, 2020-11-30:
I think it would be good to have something that happens automatically at commit time.

I also played around with trying to figure out git's pre-commit feature, and I couldn't get it to do exactly what I wanted.  Anything that uses pre-commit seems like it would require extra steps from the user to set it up before it's able to enforce style anyway.

If we're going to rely on users behaving correctly anyway, then we could just say we have a policy that everyone needs to run a style-enforcing script before they can do a PR.  Here's a little script I made that formats everything, and then tells you if it needs to be run again*:
---------------------------
#!/bin/bash

find . -name \*.h -exec clang-format --style=llvm -i {} \;
find . -name \*.cc -exec clang-format --style=llvm -i {} \;
echo "LLVM style formatting complete.  Run this again if any style violations appear below."
find . -name \*.h -exec clang-format --style=llvm --dry-run {} \;
find . -name \*.cc -exec clang-format --style=llvm --dry-run {} \;
-------------------------
*I don't know why clang-format doesn't always get everything in the first go, and sometimes needs to be run again, but that has been my experience trying it out.

closed 2022-01-29 by ngober