ChampSim/ChampSim #90: Two questions about the OoO code

alberto-ros, 2020-11-20:
Hi,

I have two questions regarding the OoO core code. 

The first one is why in main.cc, the schedule and execute calls for normal instructions and memory instructions are alternated:

             ooo_cpu[i].schedule_instruction();
	      // execute
	      ooo_cpu[i].execute_instruction();

	      ooo_cpu[i].update_rob();

	      // memory operation
	      ooo_cpu[i].schedule_memory_instruction();
	      ooo_cpu[i].execute_memory_instruction();

In the other stages, the calls go from later to earlier stage, which makes more sense to me.

The second question is why the LQ or SQ are not filled along with the ROB at dispatch, but at schedule. It seems to me that these queues can be filled out-of-order even if, as commented in the code, the schedule stage is in-order. My understanding was the ROB, LQ, and SQ are filled at the same stage, and usually at the last in-order stage at the front-end.

Thanks a lot,
Alberto

sethpugsley, 2020-11-30:
These are both good points.  Eventually someone should do something about both of them.

For the first point--I agree that we should be more consistent in the last-to-first ordering of things within the main simulation loop.  Also, the redundant calls to update_rob() should probably be eliminated.  This fix should be fairly straightforward.

To answer the second point--I didn't know what I was doing when I first coded up all of that stuff years ago.  That's why the LQ and SQ entries are allocated when they are, and why they are separate from the L1D read and write queues.  This is clearly not correct, but it is functional for now.  I'm scared of touching all the LQ and SQ stuff though, because it's pretty complicated.  Anyone else is welcome to give it a try, but it's not high on my priority list right now.

alberto-ros, 2020-12-02:
Thanks, Seth. I will take a deeper look at the OoO and see if I can do some modifications in this regard.