ChampSim/ChampSim #102: Issue with generating traces using ChampSimTracer on Pin 3.2

rajesh-s, 2020-12-11:
Hi,

I'm having trouble generating traces with the Pin version 3.2 that's documented in the README. Has anyone encountered this before? Does PIN3.2 work with recent linux kernels as well? 

I'm not sure if it's a PIN issue or if I'm missing something with the champsim_tracer.

Command:
```pin-3.2-81205-gcc-linux/pin -ifeellucky -t obj-intel64/champsim_tracer.so -- ls```

OS:
Ubuntu 20.04.1 LTS
uname -r: 5.4.0-54-generic

Output:
```
A: Source/pin/elfio/img_elf.cpp: ProcessSectionHeaders: 601: assertion failed: SEC_vaddr_i(sec) >= IMG_seg_text_vaddr_i(img) && SEC_vaddr_i(sec) < IMG_seg_data_vaddr_i(img)

NO STACK TRACE AVAILABLE
Pin 3.2
Copyright (c) 2003-2016, Intel Corporation. All rights reserved.
@CHARM-VERSION: $Rev: 81201 $
Aborted (core dumped)
```

Thanks!

rajesh-s, 2020-12-11:
Resolved this by using a newer tool version PIN3.17 from [here](https://software.intel.com/content/www/us/en/develop/articles/pin-a-binary-instrumentation-tool-downloads.html)

There are limitations with PIN3.2 on newer Ubuntu versions (that have newer linux kernels). Might run into issues like [this](https://stackoverflow.com/questions/55698095/intel-pin-tools-32-bit-processsectionheaders-560-assertion-failed)

The only change that was required to get ChampSim tracer working with PIN3.17 on Ubuntu20.04.1 LTS was the addition of use namespace std proposed in #101

closed 2020-12-11 by rajesh-s

rajesh-s, 2020-12-11:
You can test Pin tool installation/compatibility with your OS by using the following before using it with ChampSim tracer:

```bash
wget https://software.intel.com/sites/landingpage/pintool/downloads/pin-3.17-98314-g0c048d619-gcc-linux.tar.gz
tar -xzvf pin-3.17-98314-g0c048d619-gcc-linux.tar.gz
cd pin-3.17-98314-g0c048d619-gcc-linux/source/tools/ManualExamples
make
cd ../../../
./pin -t source/tools/ManualExamples/obj-intel64/inscount0.so -- ls # Testing with LS command
```

Yujie-Cui, 2021-08-24:
I executed your above command, it was successful.  After the execution, the current directory file is displayed.  

But after execute champ-sim/tracer/make_tracer.sh. 

When executing 
```
./pin-3.17-98314-g0c048d619-gcc-linux/pin -injection child -ifeellucky -t obj-intel64/champsim_tracer.so -o traces/ls_trace.champsim -s 100000 -t 200000 -- ls
E: Pin does not support instrumenting this conditional branch: [xbegin 0x7eff008048d8] with IARG_BRANCH_TAKEN error.
```