ChampSim/ChampSim #104: Address space randomization

hut95, 2020-12-18:
After my trace file generation in pintool, when I run it on champsim I do not see the address in my memory operations in champsim reflected. Why does such an address randomization happen and how to overcome this? Do you have any feature that stops address space randomization?

sethpugsley, 2020-12-18:
Are you asking about the virtual to physical translation?  Are you saying that you want the physical addresses to be equal to the virtual addresses?  Also, which branch of the code are you using, master or develop?

hut95, 2020-12-18:
I am talking about the virtual addresses that I generate using the pintool are not reflected back when I run the same on champsim. The addresses I see are randomized. Also I am using master.

sethpugsley, 2020-12-19:
By random do you mean they have no correlation with each other?  Or do you mean they are just different than you expect?  Are they really just random garbage?  Are you able to write a simple trace reading program that can determine if the problem is with the trace generation or inside ChampSim?

hut95, 2020-12-19:
I meant they're not the same as one would expect. By randomization I meant, the addresses are not what I see on pintool is not reflected back on champsim. I don't see random garbage values. All these values are surely valid just that they're randomized from the trace file's point of view. It is not an issue with the trace file for sure. I did verify on this.

sethpugsley, 2020-12-20:
I'm still struggling to understand the exact nature of the problem you're having.  How are you observing what addresses the pintool sees?  How do you know those are different from the virtual addresses that ChampSim is using?  Can you please explain, in great detail, your workflow that led you to this conclusion?  By this, I mean list things like "I changed the pintool tracer to also print out virtual addresses for all loads," or "I modified ChampSim to print out the virtual addresses of all loads," or "I wrote a simple trace reader that shows that disagrees with ChampSim's virtual addresses."  Also, make sure you're comparing hex-to-hex, or dec-to-dec when you're claiming the VAs don't match up!

hut95, 2020-12-21:
I modified a tracer on pintool to print out all virtual addresses of source and destination addresses for a simple memcpy operation. I also printed out all the virtual addresses in champsim for the load and store operation. I am comparing only hex-to-hex in my case. I observe that the set of source and destination addresses that I print using the tracer file in pintool, is no where to be found on champsim printed addresses. I do not conclude that they are getting randomized, I just wanted to know why they may be getting randomized because from my earlier experience of running trace files on such simulators, I do find the reflection of my addresses from trace files on the simulator dump as well.

sethpugsley, 2020-12-21:
Is your memcpy() between dynamically- or statically-allocated buffers?  Do you see the exact same virtual addresses across many runs of your memcpy() program?

hut95, 2020-12-21:
No the addresses change for each time I run my memcpy() and I can confirm I generate fresh trace files in pintool everytime I re-run my program to generate addresses.

ngober, 2022-05-12:
I have seen this happen, where the same program generates different virtual addresses on different runs of PIN. This appears to be a limitation of PIN. The addresses seem to have a different base on separate runs, but their offsets appear to be constant. This shouldn't make a huge impact on ChampSim.

closed 2022-05-12 by ngober