ChampSim/ChampSim #105: Inconsistent Prefetcher Parameters

manaal22152, 2020-12-20:
Why is the sum of useful and useless prefetches not equal to the total prefetches issued? Moreover, what is the difference between 
PREFETCH  ACCESS and PREFETCH  REQUESTED? I have attached a log file for reference. In the log file, the sum of useful and useless prefetches is not equal to the prefetches issued for L1D and L2C. For LLC, prefetches issued are zero, but useful and useless prefetches are non zero quantities. How is this possible?
[astar_23B.trace.xz-bimodal-no-bip-bip-no-lru-1core.txt](https://github.com/ChampSim/ChampSim/files/5720853/astar_23B.trace.xz-bimodal-no-bip-bip-no-lru-1core.txt)

GinoAC, 2020-12-20:
The sum isn't equal because some prefetches are in the cache and are not evicted or accessed before the simulation ends, making them not count towards the useful or useless metric. 

PREFETCH ACCESSED is the number of prefetches handled by the cache, including prefetches that are being requested by another level of the cache hierarchy. PREFETCH_REQUESTED is the number of prefetches that are successfully placed in the prefetch queue from that cache level's prefetcher. 

ChampSim's cache is non-inclusive, meaning a prefetch request from the L1 will fill both the L2 and L3. The useful and useless metric tracks any prefetch in the cache, even if it didn't originate from that level of the cache.

manaal22152, 2020-12-21:
So what is the standard formula to find coverage (compared to no prefetcher) and accuracy of a prefetcher from the log files? I have tried doing accuracy=USEFUL/ISSUED, but that gives a very low value for the prefetchers which have shown a very high accuracy in their research papers using ChampSim.

ngober, 2020-12-23:
I usually use accuracy=useful/(useful+useless) in my calculations. I suspect this still underestimates the accuracy, since it doesn't take into account prefetches that were issued, but not marked useful due to the end of the simulation. Does this help?

manaal22152, 2020-12-24:
And what formula to use for finding coverage?

GinoAC, 2020-12-24:
I use `accuracy=useful/issued` and assume that anything not counted as useful by the end of the simulation is useless. A prefetch not used by the end of a simulation did not provide any benefit and possibly evicted a useful cache line, effectively making it a useless prefetch. If you're trying to find the accuracy in the L3 with an L2 prefetcher, use the number of prefetch misses seen by the L3 instead of ISSUED, since that's also the number of prefetches that were placed in the L3. 

For coverage I use `(baseline_load_misses - test_load_misses)/(baseline_load_misses)`. Where test_load_misses is from whatever new configuration you're testing. I'd recommend using the number of load misses since they're are on the critical path and an aggressive prefetcher will keep the total number of misses fairly consistent since its ideally trying to replace critical misses with timely prefetch misses.

manaal22152, 2020-12-24:
I am confused. @ngober mentioned accuracy=useful/(useful+useless) and @GinoAC  mentioned accuracy=useful/issued. The two formulas give very different answers for all the prefetchers I have tested. The former gives accuracy in the range of 80-90% whereas the latter reduces it to meagre 10-20%. I think the first formula is correct as the results match with the papers I have read.

GinoAC, 2020-12-24:
Its odd that there would be such a large disparity between the two accuracy measurements across most benchmarks if you're measuring a standalone prefetcher. I notice you're working off the main branch and based on the code and the results you've posted, it looks like the prefetch statistics are not being reset after a warmup.  PR #109 fixes this issue.

ngober, 2020-12-25:
Accuracy and coverage in prefetching work are actually pretty poorly-defined metrics. I recommend always being clear in your publications about how you are calculating them.

When I calculate coverage, I typically use `useful/(useful + useless + load_miss + rfo_miss)`. This differs from Gino's and will consistently underestimate the coverage, but I feel it more clearly captures the effect that a prefetcher may have causing premature evictions of non-prefetched lines.

That discussion may stray too far from the root of this issue, though. The question is whether it is more clear to include in our counts prefetches filled but not known to be useful or useless. I think we should keep them, since this may give information about overprefetching. I could be convinced, though, that prefetch misses in a higher level should not be counted in the prefetches for the lower level.

manaal22152, 2020-12-25:
What I conclude is that there is no fixed formula. For the triage prefetcher that I will be using as a baseline, they have used accuracy=useful/(useful+useless) and coverage=useful/(useful + load_miss + rfo_miss). I will follow the same. I have final two questions:
1. I have a prefetcher at L1 and L2. My baseline prefetcher works only at L2. In order to compare the two prefetchers, should I compare the accuracy and coverage at all the three levels(L1, L2 and LLC)? Or should I compare the average of these parameters of the former prefetcher at L1 and L2 to the value of the latter at L2? Or is there some other way?
2. What about multicore results? Should I take average of all 4 cores?

ngober, 2020-12-27:
In general, you should always be clear in your publications how you calculate composite metrics like accuracy and coverage. This allows others to understand the claims you make.

ChampSim's purpose is not to report composite metrics. When ChampSim performs its simulation, it calculates only counting metrics, such as number of hits and misses. In my opinion, it is the responsibility of the researcher, not the tool, to interpret counting metrics for their audience.

I say this because it is not ChampSim's place to tell you how to compare metrics at different levels of the cache or in multicore situations. You should survey the literature and report your metrics in a way that your audience will be able to understand and that communicates your findings correctly. I would caution you that your intuition may be dangerous, since taking the mean of percentages without accounting for the population size is ripe for catastrophic errors.

manaal22152, 2020-12-28:
I did survey the literature, but none of the papers which have used ChampSim have mentioned how they have estimated the accuracy and coverage of their prefetcher. Since ChampSim is also used for DPC, I thought there might be a standard way of doing it. Thank a lot @ngober  and @GinoAC for the clarifications.

closed 2020-12-28 by manaal22152