ChampSim/ChampSim #108: Fetcher issues (develop branch)

lpha-z, 2020-12-24:
I changed source code a little, then I compiled by default configuration and ran ChampSim with the following command: `bin/champsim -warmup_instructions 0 -simulation_instructions 2500 -traces client_001.champsimtrace.xz`
Then I found some fetcher issues.

```diff
--- a/inc/champsim.h
+++ b/inc/champsim.h
@@ -6,7 +6,7 @@
 #include "champsim_constants.h"

 // USEFUL MACROS
-//#define DEBUG_PRINT
+#define DEBUG_PRINT
 #define SANITY_CHECK
 #define LLC_BYPASS
 #define DRC_BYPASS

--- a/src/ooo_cpu.cc
+++ b/src/ooo_cpu.cc
@@ -298,7 +298,7 @@ uint32_t O3_CPU::check_rob(uint64_t instr_id)

     for (uint32_t i=ROB.head, count=0; count<ROB.occupancy; i=(i+1==ROB.SIZE) ? 0 : i+1, count++) {
         if (ROB.entry[i].instr_id == instr_id) {
-            DP ( if (warmup_complete[ROB.cpu]) {
+            DP ( if (warmup_complete[cpu]) {
             cout << "[ROB] " << __func__ << " same instr_id: " << ROB.entry[i].instr_id;
             cout << " rob_index: " << i << endl; });
             return i;
@@ -401,7 +401,7 @@ void O3_CPU::fetch_instruction()
          else
               trace_packet.address = ifb_entry.ip >> LOG2_PAGE_SIZE;
           trace_packet.full_addr = ifb_entry.ip;
-         trace_packet.instr_id = 0;
+         trace_packet.instr_id = ifb_entry.instr_id;
          trace_packet.rob_index = i;
           trace_packet.ip = ifb_entry.ip;
          trace_packet.type = LOAD;
@@ -450,7 +450,7 @@ void O3_CPU::fetch_instruction()
           fetch_packet.full_addr = ifb_entry.instruction_pa;
           fetch_packet.v_address = ifb_entry.ip >> LOG2_PAGE_SIZE;
           fetch_packet.full_v_addr = ifb_entry.ip;
-         fetch_packet.instr_id = 0;
+         fetch_packet.instr_id = ifb_entry.instr_id;
          fetch_packet.rob_index = 0;
           fetch_packet.ip = ifb_entry.ip;
          fetch_packet.type = LOAD;
```

### ITLB is accessed multiple times but is not needed.

```
[ITLB_RQ] add_rq instr_id: 2385 address: ffffa844d full_addr: ffffa844dacc type: 0 head: 9 tail: 10 occupancy: 1 event: 24176 current: 24175
[ITLB_RQ] add_rq instr_id: 2391 address: ffffa844d full_addr: ffffa844dae4 type: 0 head: 10 tail: 11 occupancy: 1 event: 24177 current: 24176
[ITLB_RQ] add_rq instr_id: 2391 address: ffffa844d full_addr: ffffa844dae4 type: 0 head: 11 tail: 12 occupancy: 1 event: 24178 current: 24177
[ITLB_RQ] add_rq instr_id: 2397 address: ffffa844d full_addr: ffffa844dafc type: 0 head: 12 tail: 13 occupancy: 1 event: 24179 current: 24178
```

In cycle 24175, an ITLB request is issued by instr2385.
In cycle 24176, another ITLB request is issued by instr2391.
In cycle 24177, the first ITLB request is returned, and the following code restores `instr2391.translated == INFLIGHT` to `instr2391.translated == 0`, then instr2391 issues yet another ITLB request since `instr2391.translated == 0`.
In cycle 24178, the second ITLB request is returned, and it sets `instr2391.translated == COMPLETED`.
In cycle 24179, the third ITLB request is returned, but no one uses it.

https://github.com/ChampSim/ChampSim/blob/37fc62280c0752dd7c8c6f9a981c1833cf4fd4bb/src/ooo_cpu.cc#L1698-L1699

The root of this problem is that the following codes assume "all instructions on the same line can be translated in one cycle" while the above code assumes "only 6 instructions can be translated in one cycle".

https://github.com/ChampSim/ChampSim/blob/37fc62280c0752dd7c8c6f9a981c1833cf4fd4bb/src/ooo_cpu.cc#L417-L437

### Instruction fetch is done out of order. 

```
[L1I_RQ] add_rq instr_id: 2391 address: 2f0b6b full_addr: bc2dae4 type: 0 head: 3 tail: 4 occupancy: 1 event: 24182 current: 24178
[L1I_RQ] add_rq instr_id: 2398 address: 2f0b6c full_addr: bc2db00 type: 0 head: 3 tail: 5 occupancy: 2 event: 24183 current: 24179
[L1I_RQ] add_rq instr_id: 2408 address: 2f0b6f full_addr: bc2dbe4 type: 0 head: 3 tail: 6 occupancy: 3 event: 24184 current: 24180
[L1I_RQ] add_rq instr_id: 2415 address: 2f0b70 full_addr: bc2dc00 type: 0 head: 4 tail: 7 occupancy: 3 event: 24186 current: 24182
[L1I_RQ] add_rq instr_id: 2389 address: 2f0b6b full_addr: bc2dadc type: 0 head: 5 tail: 8 occupancy: 3 event: 24187 current: 24183
[L1I_RQ] add_rq instr_id: 2404 address: 2f0b6c full_addr: bc2db18 type: 0 head: 6 tail: 9 occupancy: 3 event: 24188 current: 24184
[L1I_RQ] add_rq instr_id: 2431 address: 2f0b71 full_addr: bc2dc40 type: 0 head: 6 tail: 10 occupancy: 4 event: 24188 current: 24184
[L1I_RQ] add_rq instr_id: 2414 address: 2f0b6f full_addr: bc2dbfc type: 0 head: 6 tail: 11 occupancy: 5 event: 24189 current: 24185
[L1I_RQ] add_rq instr_id: 2438 address: 19e74ee full_addr: 679d3b90 type: 0 head: 7 tail: 12 occupancy: 5 event: 24190 current: 24186
[L1I_RQ] add_rq instr_id: 2421 address: 2f0b70 full_addr: bc2dc18 type: 0 head: 8 tail: 13 occupancy: 5 event: 24191 current: 24187
[L1I_RQ] add_rq instr_id: 2450 address: 19e74ef full_addr: 679d3bc0 type: 0 head: 10 tail: 14 occupancy: 4 event: 24192 current: 24188
[L1I_RQ] add_rq instr_id: 2437 address: 2f0b71 full_addr: bc2dc58 type: 0 head: 11 tail: 15 occupancy: 4 event: 24193 current: 24189
[L1I_RQ] add_rq instr_id: 2466 address: 19e74f0 full_addr: 679d3c00 type: 0 head: 12 tail: 16 occupancy: 4 event: 24194 current: 24190
[L1I_RQ] add_rq instr_id: 2444 address: 19e74ee full_addr: 679d3ba8 type: 0 head: 13 tail: 17 occupancy: 4 event: 24195 current: 24191
[L1I_RQ] add_rq instr_id: 2427 address: 2f0b70 full_addr: bc2dc30 type: 0 head: 14 tail: 18 occupancy: 4 event: 24196 current: 24192
```

The same thing is happening.
https://github.com/ChampSim/ChampSim/blob/37fc62280c0752dd7c8c6f9a981c1833cf4fd4bb/src/ooo_cpu.cc#L1744-L1745

sethpugsley, 2021-01-10:
Hmmmmmm, this is tricky.  I think we should start out by clearly defining what we believe the correct behavior should be.

We could remove all parallelism and force the first instruction read to be completely translated and fetched before the second one begins.  That would be slow, but would eliminate this behavior.

What can we do that would allow for parallel outstanding requests, but also force everything to be in-order?  The current out-of-orderness comes from the re-tries, right?  I guess that means that we should only allow an instruction to issue its ITLB request, or its L1I request if we're 100% positive that it will not need to retry.  Can anyone think of a way of doing that w/o doing the trivial serialization method mentioned above?

ngober, 2021-01-11:
I'm looking carefully at the example given about the ITLB. In cycle 24175, I suspect that instr2391 is not in the fetch buffer yet, so it is not marked as inflight, though it shares a page with instr2385. So, a request for it is issued after it enters the fetch buffer in cycle 24176. This request is "cancelled" in cycle 24177 because it does not fit in the bandwidth of the prior request, but the core cannot determine that instr2391 corresponds to a different TLB request.

The current implementation assumes that all instructions currently in the fetch buffer that could correspond to a particular request do. This effect is particularly pronounced for the L1I, since it has a larger latency. My initial reaction is to attach a set of iterators to the packet that point to instructions in the fetch buffer that are waiting on this packet. We already do something like this with the backend:

https://github.com/ChampSim/ChampSim/blob/37fc62280c0752dd7c8c6f9a981c1833cf4fd4bb/inc/block.h#L35

We could replicate this functionality for the fetch mechanism.

ngober, 2021-01-27:
Resolved by #120.

PR #120 (2021-01-27): Prevented repeated fetches to the same blocks.
Resolves #108.

This patch tightens the restrictions on how fetch (and fetch translate) requests may be fulfilled. The L1I and ITLB can now issue requests that satisfy larger numbers of instructions, up to the fetch buffer's capacity, but those requests must be contiguous in program order. I do not allow a block of instructions to issue a request if the tail of the block hits the end of the fetch buffer, since there may be instructions still to come within the block (see discussion in #108). The only exception to this is when the entire fetch buff [... cut]
closed 2021-01-27 by ngober