ChampSim/ChampSim #111: Run single workload on multicore

manaal22152, 2020-12-30:
I want to run a single workload on 4 core configuration. Is it possible using ChampSim?

ngober, 2021-01-04:
Have you tried running a 4-core configuration with four copies of the same trace?

manaal22152, 2021-01-05:
That would be running 4 workloads on 4-cores. I want to run a single workload on 4 -core for calculation of weighted IPC.

ngober, 2021-01-05:
ChampSim, being a trace-based simulator, needs one trace per core. Otherwise, cores will sit entirely idle while the others run. If you were to run a single core in a 4-core simulation, it would be no different than a 1-core simulation. Currently, ChampSim fails to run if fewer traces than cores are given.

manaal22152, 2021-01-18:
Thanks a lot!

closed 2021-01-18 by manaal22152