ChampSim/ChampSim #118: -nan average miss latency

ArnauPrat, 2021-01-22:
Hi all, I have the following output obtained from the 602.gcc_s-1850B.champsimtrace.xz trace.
It reports an average "L1I AVERAGE MISS LATENCY: -nan cycles latency" 

It looks like a bug to me. Is that correct?

Thanks

[602.gcc_s-1850B.champsimtrace.xz-bimodal-no-no-no-no-lru-1core.txt](https://github.com/ChampSim/ChampSim/files/5854858/602.gcc_s-1850B.champsimtrace.xz-bimodal-no-no-no-no-lru-1core.txt)

itisntalex, 2021-01-22:
You trace specifies that you have no misses in the L1I which means that you're dividing by 0. That's why.

ArnauPrat, 2021-01-22:
I Understand. Thanks for your prompt response.

ngober, 2021-01-22:
I'm closing this because it appears to be resolved. If this is not the case, let me know and I can reopen it for further discussion.

closed 2021-01-22 by ngober