ChampSim/ChampSim #119: Instruction pointer usage

hut95, 2021-01-25:
Hi,
Which variable in the ooo_cpu code points us to the IPs seen in the object dump of a particular binary? I see there are multiple IPs defined and print statements are not very specific in this regard.

Thanks

ngober, 2021-01-27:
Each instruction in the core knows its own IP. If you wanted to know what IPs are seen, you can collect that data at the front of the core, in `O3_CPU::init_instruction()` or at the back in `O3_CPU::retire_rob()`. But, if all you want to do is see the IP, ChampSim may not be the tool you're looking for: you could simply read the trace file. Its format is in `inc/instruction.h`.

hut95, 2021-01-27:
So I want to make use of IP that I see from my object dump for a binary and set a flag in champsim based on the instruction pointer identification to simulate a particular region of code trace. I saw that there are multiple IPs in the print statements but I do not see the IPs I see in my objectdump on champsim. I don't just want to see the IP but I want to search for it and use it to set a flag in champsim.

hut95, 2021-02-01:
Any follow up on this? Any help would be appreciated. 

Thanks

sethpugsley, 2021-02-01:
Are the IPs you're interested in inside system calls?  This could explain why you don't see them, because PIN can't trace anything except user code.  Also, have you ruled out [address space layout randomization](https://en.wikipedia.org/wiki/Address_space_layout_randomization)?  Do you see the same IPs every time you run your program?  Are the IPs you see in ChampSim in the same ballpark as the IPs you're looking for from your object dump, or are they way off?

hut95, 2021-02-01:
I see the same IPs everytime I run the program but the IPs I see in champsim are way of the ones that I see from the objectdump. Earlier when I ran smaller sized memcpy transfers I was able to see IPs from the object dump but now as my memcpy size has increased I no longer see any IPs from my binary's object dump. I have a work around in place for the address space issue. The IPs that I am interested in are the IPs of my memcpy transaction which is a syscall in itself.

hut95, 2021-02-04:
Any further inputs on this would be appreciated. Thanks.

sethpugsley, 2021-02-04:
Are you seeing any bits of IP where you expect them to be?  Are all of the IPs offset by a constant?  Does it look totally random?

sethpugsley, 2021-02-04:
I went back to the Pin tool documentation, and looked at the example code they have, and I think you should look at source/tools/ManualExamples/itrace.cpp in your Pin installation.  Compare and contrast how this official sample code works directly with VOID* data types, versus how the ChampSim tracer converts them to unsigned long long int format.  There could be something to this.  I really have no idea, I'm just throwing something out there.

You might consider trying to change the ChampSim tracer so that it never does this type casting, and modify the trace_instr_format data type to use VOID* directly instead of unsigned long long int for all address type members.  If you do this, you should double and triple check that VOID* and unsigned long long int are the same number of bytes on your platform.  I DON'T imagine you would need to make a corresponding change to ChampSim itself, just the tracer.

hut95, 2021-02-04:
Thanks a lot for your help. I will take a look into the tracer.

ngober, 2021-02-05:
As a note, if you find something wrong with the tracer, we would welcome any pull requests that fix bugs in it. Only a minority of ChampSim users use the tracer directly. I know of only one other person who is actively working with it.

closed 2021-03-16 by ngober