ChampSim/ChampSim #122: Add TAGE branch predictor

AminAvan, 2021-01-27:
I want to add "[TAGE](https://github.com/2Bor2C/tage)" branch predictor on ChampSim, and I read the README in ChampSim to how to add our branch predictor ($ cp branch/branch_predictor.cc prefetcher/mybranch.bpred), but:
1) I think the "prefetcher" should replace with "branch"
2) It is not sufficient for me to add TAGE on ChampSim
Thanks

ngober, 2021-01-27:
Your first point is correct. That looks like a typo and I've corrected it in master.

I'm not sure what your second point is saying. If you don't see `branch/branch_predictor.cc`, you may need to build ChampSim first. In the master branch, this is a generated file, and you might not have run the script to generate it.

AminAvan, 2021-01-27:
Thank you,
For the second point, when I use `branch/branch_predictor.cc branch/TAGE.bpred` to build a [TAGE](https://github.com/2Bor2C/tage) branch predictor, but I cannot edit\join correctly the TAGE code in ChampSim, to build TAGE correctly
Bests

AminAvan, 2021-01-27:
I mean each branch predictor has its own features that correctly connect to the ChampSim simulator, how can I add TAGE branch predictor correctly? What steps should I take?
Bests

ngober, 2021-01-27:
I looked through the code you linked, and it looks to be a very similar interface to ChampSim (I think it's the CBP5 simulator?). Both have two functions: one to get the prediction and another to update the predictor. I would say you should pass the information between these interfaces and it should work.

AminAvan, 2021-01-28:
As I am new in working with ChampSim, would you please be a mentor for me on the project?

ngober, 2021-01-28:
I appreciate the ask, but since I am a maintainer of this simulator in addition to a researcher myself, I can't commit to guiding you on your project. You might consider asking the ChampSim mailing list (linked in the README). That is a better venue for a discussion like this. You may even be able to find a native ChampSim implementation.

closed 2021-01-31 by AminAvan