ChampSim/ChampSim #130: Questions on the cache prefetching model in ChampSim

itisntalex, 2021-02-17:
Good evening (morning?) folks!

I am writing here as a couple of questions about how cache prefetching is actually implemented in ChampSim came up to me. Here they are:

- As far as I understood going through the code of caches, ChampSim does not distinguish demand and prefetch access as a source of subsequent prefetch request. Is that correct? I believe that only demand accesses should be used to trigger prefetch request...
-  The second is about a piece of code that I came across in the`CACHE::handle_prefetch` method. ChampSim allows prefetchers to throw prefetch request over prefetch request coming from higher caches. Is that realistic?

Thanks in advance! 😄

ngober, 2021-02-17:
I feel like both of your questions are about the same thing, so I will attempt to answer them together. ChampSim does distinguish between demands and prefetches. This information is accessible in the packet's `type` member, which propagates with a packet as it traverses the cache hierarchy. This information is passed on to the prefetchers (as a parameter in the `l1d_prefetcher_operate()` function, for example).

The prefetcher operation functions are called only for load packets as well as prefetches from higher levels. From what I have seen in the state of the art, this appears to be realistic. But, if your research would suggest that it is not, you can design your prefetcher to do nothing when the type is prefetch.

This brings up a point, however, that RFO packets appear never to be acted on, whether they hit or miss and regardless of the level of the cache (they are writes in the L1, and reads everywhere else). Writeback packets are never acted upon, and this may be realistic, but I suppose that it may be better to adopt a freer approach and act on every packet, then rely on the user to ignore irrelevant packets.

itisntalex, 2021-02-18:
First of all, thank you for taking the time to answer my question.

I would like to make sure of one thing and also ask your opinion on the best approach to take on handling prefetches in ChampSim.

- First of all, I believe that the answer you gave referred mostly to what is available in the develop branch. However, I was pointing at what is available in the master branch as I believe it's the one most of us should rely on for working purposes.
- That brings me to my second point which is about the two approaches taken by these branches. The question is rather simple: if you had to choose between those two approaches on implementing prefetching in ChampSim, which one would you pick?

itisntalex, 2021-02-18:
First of all, thank you for taking to answer my question.

However, I do want to make a point as I believe you answered considering the `develop` branch whereas I was considering the `master`. I believe that's the branch we should refer to by default as it is the one most of should be using for work/research purposes. Here is my question:

- Knowing that there are currently two approaches to prefetching implemented in the repository - one per branch -, which one were you actually referring to. Also, if you had to pick one of them, which one do you think is actually the most relevant of them?

ngober, 2021-02-18:
I might be mistaken, but I believe that the master and develop branch are the same in this regard. It is harder to read in the master branch, but the prefetcher function calls are gated behind if-statements that check for being a load packet or an upper-level prefetch packet.

closed 2021-05-24 by ngober