ChampSim/ChampSim #131: Simulating different page sizes

gvavou5, 2021-02-18:
Hello, 

I would like to see the impact of different page sizes. To do so, I did the following experiments: (i) page size is 4KB, (ii) page size is 2MB. Note that **both experiments don't use prefetching at any cache level** and when I consider 2MB pages I properly changed both PAGE_SIZE and LOG2_PAGE_SIZE in inc/champsim.h. 

Intuitively, 2MB pages would only improve performance by saving TLB misses (i.e., page walks) because the current version of ChampSim **does not have** a page table walker that stores entries from all page table levels into the caches. Hence, the cache MPKI rates should be the same for both experiments explained above.

My evaluation verified that there is no change in the cache MPKI rates when using different page sizes (intuitively correct) but I observed that when using 2MB pages the average cache latencies are much higher than the ones reported when 4KB pages are used. 
For instance, the L2C average latency of trace 410.bwaves_1963 is 296cc and 565cc when using 4KB and 2MB pages, respectively.
I don't see why such behavior is normal. The average latencies should be the same. Am I wrong? Does anyone know anything about it?

ngober, 2021-08-12:
This appears to be resolved in #172.

I ran a similar test to you. I ran 410.bwaves-1963B for 10M/100M and I see very similar latencies and MPKIs for all of the data caches.

ngober, 2022-01-21:
Resolving after the merge of #172.

closed 2022-01-21 by ngober