ChampSim/ChampSim #133: Deadlock Observed for Some of the Traces

spikemar, 2021-02-25:
Hello, 

When using any combination of the prefetchers/branch_predictor/replacement_policy in the many-core simulations I am getting a deadlock for a subset of the traces. 

Using bimodal-no-no-no-no-lru-4core and running for 200M warmup + 1B on four instances of 605.mcf_s-1644B.champsimtrace.xz, gives me deadlock. I've tried running on both the master and the developer branches and observe the same issue in both, although as a difference, the developer branch gives deadlock much more quickly for this trace. 

Is this a known issue? 

Full list of traces giving me issues:
654.roms_s-293B.champsimtrace.xz
602.gcc_s-1850B.champsimtrace.xz
605.mcf_s-1536B.champsimtrace.xz
603.bwaves_s-2609B.champsimtrace.xz
619.lbm_s-3766B.champsimtrace.xz
649.fotonik3d_s-10881B.champsimtrace.xz
429.mcf-184B.champsimtrace.xz
429.mcf-22B.champsimtrace.xz
605.mcf_s-994B.champsimtrace.xz
603.bwaves_s-1740B.champsimtrace.xz
429.mcf-51B.champsimtrace.xz
619.lbm_s-2677B.champsimtrace.xz
603.bwaves_s-891B.champsimtrace.xz
429.mcf-192B.champsimtrace.xz
605.mcf_s-1644B.champsimtrace.xz
605.mcf_s-484B.champsimtrace.xz
619.lbm_s-2676B.champsimtrace.xz
619.lbm_s-4268B.champsimtrace.xz

ngober, 2021-02-25:
I've seen this issue come up a few times and I've never been able to track it down. Can you see anything in the deadlock output that might point us in the right direction? For instance, is the head of the ROB not being scheduled? Maybe a memory request packet is not being returned up the hierarchy correctly?

spikemar, 2021-02-26:
Hello, 

Unfortunately, I don't have a great amount of domain knowledge in this field. I'm pasting the output associated with 429.mcf-184B.champsimtrace.xz maybe that will be somewhat helpful. 

From my understanding, are you saying that the bug is not reproducible on your end?

Heartbeat CPU 0 instructions: 520000001 cycles: 5352856104 heartbeat IPC: 0.0574084 cumulative IPC: 0.0604053 (Simulation time: 2 hr 28 min 33 sec) 
Heartbeat CPU 1 instructions: 520000001 cycles: 5352937633 heartbeat IPC: 0.0575625 cumulative IPC: 0.0605552 (Simulation time: 2 hr 28 min 33 sec) 
Heartbeat CPU 2 instructions: 520000001 cycles: 5358046750 heartbeat IPC: 0.0572812 cumulative IPC: 0.0606074 (Simulation time: 2 hr 28 min 40 sec) 
Heartbeat CPU 3 instructions: 520000001 cycles: 5404530042 heartbeat IPC: 0.0569069 cumulative IPC: 0.0601362 (Simulation time: 2 hr 29 min 43 sec) 
DEADLOCK! CPU 3 instr_id: 525342047 translated: 2 fetched: 2 scheduled: 2 executed: 1 is_memory: 1 event: 5814321728 current: 5815321728

Load Queue Entry
[LQ] entry: 0 instr_id: 525342047 address: 661beeac0028 translated: 2 fetched: 0
[LQ] entry: 1 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 2 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 3 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 4 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 5 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 6 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 7 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 8 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 9 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 10 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 11 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 12 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 13 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 14 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 15 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 16 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 17 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 18 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 19 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 20 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 21 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 22 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 23 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 24 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 25 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 26 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 27 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 28 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 29 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 30 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 31 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 32 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 33 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 34 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 35 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 36 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 37 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 38 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 39 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 40 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 41 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 42 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 43 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 44 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 45 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 46 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 47 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 48 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 49 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 50 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 51 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 52 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 53 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 54 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 55 instr_id: 0 address: 0 translated: 0 fetched: 0
[LQ] entry: 56 instr_id: 0 address: 0 translated: 0 fetched: 0
[... 139 lines cut ...]
[SQ] entry: 66 instr_id: 0 address: 0 translated: 0 fetched: 0
[SQ] entry: 67 instr_id: 0 address: 0 translated: 0 fetched: 0
[SQ] entry: 68 instr_id: 0 address: 0 translated: 0 fetched: 0
[SQ] entry: 69 instr_id: 0 address: 0 translated: 0 fetched: 0
[SQ] entry: 70 instr_id: 0 address: 0 translated: 0 fetched: 0
[SQ] entry: 71 instr_id: 0 address: 0 translated: 0 fetched: 0

L1D_MSHR Entry
[L1D_MSHR] entry: 0 instr_id: 525342047 rob_index: 351 address: 1986fbab000 full_addr: 661beeac0028 type: 0 fill_level: 1 lq_index: 0 sq_index: 0
[L1D_MSHR] entry: 1 instr_id: 0 rob_index: -1 address: 0 full_addr: 0 type: 0 fill_level: -1 lq_index: 0 sq_index: 0
[L1D_MSHR] entry: 2 instr_id: 0 rob_index: -1 address: 0 full_addr: 0 type: 0 fill_level: -1 lq_index: 0 sq_index: 0
[... cut ...]
[L1D_MSHR] entry: 12 instr_id: 0 rob_index: -1 address: 0 full_addr: 0 type: 0 fill_level: -1 lq_index: 0 sq_index: 0
[L1D_MSHR] entry: 13 instr_id: 0 rob_index: -1 address: 0 full_addr: 0 type: 0 fill_level: -1 lq_index: 0 sq_index: 0
[L1D_MSHR] entry: 14 instr_id: 0 rob_index: -1 address: 0 full_addr: 0 type: 0 fill_level: -1 lq_index: 0 sq_index: 0
[L1D_MSHR] entry: 15 instr_id: 0 rob_index: -1 address: 0 full_addr: 0 type: 0 fill_level: -1 lq_index: 0 sq_index: 0
bimodal-no-no-no-no-lru-4core: src/main.cc:267: void print_deadlock(uint32_t): Assertion `0' failed.

ngober, 2021-02-26:
I've previously seen this bug and at least one person has reported it on the email list. Every time I see it, though, it looks a bit different. Here, I suspect that the problem is somewhere in a shared cache, since the MSHR is allocated in the L1D, but never filled. What was the command you used to produce this output?

spikemar, 2021-02-26:
I'm using the specified command from the master branch but I have also tried the develop branch version and run the command using the format specified in that branch.

run_4core.sh bimodal-no-no-no-no-lru-4core 200 1005 4 429.mcf-184B.champsimtrace.xz 429.mcf-184B.champsimtrace.xz 429.mcf-184B.champsimtrace.xz 429.mcf-184B.champsimtrace.xz

ngober, 2021-02-26:
Could you try running a different trace on each core? Try using traces you know to be good for the first three, and see if the crash still occurs. Running the same trace on each core is an atypical use of ChampSim, since it should have entirely identical operations in all levels of the cache.

spikemar, 2021-02-26:
HI, 

I think that is fine. I've found mixed groups of traces that are still problematic but in general mixing the traces seems less error prone. Is the error that occurs here an issue that could cause performance problems for other experiments?

ngober, 2021-02-26:
I can't be certain. This has been an elusive bug and any information we can learn about it will be helpful.

spikemar, 2021-02-26:
I'll check out the mixed traces and paste group of traces any that I find with this issue (if that is helpful). Is the above repetitive set of traces useful for tracking down the issue or not so much?

spikemar, 2021-03-03:
Hello, 

The above traces will give a deadlock even when used in combination with other traces. I've actually tracked down the bug to the GCC version. Currently, I have been able to get the traces to work on gcc (ubuntu 7.5.0-3ubuntu1-18.04) 7.5.0. I haven't been able to get the binary to work correctly when compiled using CentOS ( I tried version gcc 7.4.0 and gcc 9.3.0). I'll update here after confirming:

(1) If the performance changes based on the binary.
(2) If all the above traces work.
(3) If I can find a version of gcc that will work for CentOS or if there are some other packets that are required. 

Maybe I can merge a section into the README for this issue?

ngober, 2021-03-03:
This is very interesting. I'm tracking a separate deadlock bug myself that I've suspected is a compiler/environment issue, so if the multicore bug is compiler-related, it may reflect a larger bug that is important to fix.

Am I understanding you correctly that GCC 7.4 and GCC 9.3 crash, but GCC 7.5 works, or do you mean that GCC 7.5 on Ubuntu works, but GCC 7.4 and GCC 9.3 on CentOS crashes? In other words, does the crash appear to be dependent on the compiler or the operating system?

spikemar, 2021-03-03:
Hello, 

Your 2nd understanding is correct. GCC 7.5 works on Ubuntu, GCC 7.4 and GCC 9.3 crashes on CentOS. I'm also working with IT officials at my school, they've confirmed that a static build with clang++ 9.0.1 also produces the same issue. 

I'm going to try a static build on Ubuntu and try running all the problematic traces. I'm also observed that in 8 core the issue increases to 28 different traces.

ngober, 2021-03-03:
Do you experience any of these crashes in single-core simulations? I have a build on Red Hat 7.4 that crashes with certain traces and I'm trying to figure out if these bugs could be related. Also, oddly, I can replicate your bug exactly on Ubuntu 18.04 with GCC 7.5.

spikemar, 2021-03-04:
As an update, I was actually testing running a single problematic trace at a time to narrow down the problem. Ubuntu gcc 7.5.0 helped somewhat (The single trace case runs much longer compared to CentOS) but 4 copies of the same fotonik trace is still crashing very quickly. I don't get any issues with single core as far as I can see. There seems to be some dependency on the GCC and OS version. Is there any system that you have observed to have no issues with 4C?

ngober, 2021-03-04:
I don't know that we can say yes or know for certain on that. Most of my work has been in single-core simulation (and that probably reflects in my having missed a bug like this). 

I think it's clear that there is some dependency on compiler and OS, but it's not clear to me what or how they interact.

I'm pinging @sethpugsley and @GinoAC for reinforcements. If either of you have cycles to spare, could you weigh in on this?

spikemar, 2021-03-05:
Hello, 

An update on this front, if I switch to the develop branch and compile using ubuntu 7.5.0, I think I get true deadlock instead of the issue that exists in the master brnach. After switching to the develop branch, my first problem was running out of virtual memory space. I increased that by 10X (just to make sure), and now the issue is with 1 trace in 4C and 9 traces in 8C. I think the deadlock I observe here is related to the architecture since I actually can see some of the queues filled up. Comparing the two branches might point towards whatever the issue is in the master branch. I suspect that the makefile might be part of the problem.

ngober, 2021-03-09:
I'm checking in here to let everyone know I'm still looking at this. I think the virtual memory could be part of the problem here, since it's the only place I haven't been combing over, but I don't think it's the whole issue. I have been running tests with Valgrind, and I can't find any memory errors on the development branch, so I don't think that's it. I've seen some behavior that suggests the issue is in the DRAM model, but I refactored it to trim it down and haven't seen any change in the errors. I would share it here, but I based my changes off of #124, so the diff is messy. I still can't pin down what the problem is.

I reached out to someone that I had seen having a similar error before, and he said he would also look into it.

spikemar, 2021-03-09:
Thanks for the help. I've run some additional experiments that show that the problem isn't directly related to a given set of problematic traces but has a high correlation with them. 

The following mixed sets that contain 619.lbm_s-2677B.champsimtrace.xz gave deadlock:
444.namd-33B.champsimtrace.xz
434.zeusmp-10B.champsimtrace.xz
619.lbm_s-2677B.champsimtrace.xz
649.fotonik3d_s-7084B.champsimtrace.xz

444.namd-33B.champsimtrace.xz
619.lbm_s-2677B.champsimtrace.xz
605.mcf_s-782B.champsimtrace.xz
482.sphinx3-1297B.champsimtrace.xz

The following one that still contains the same trace doesn't give deadlock:

473.astar-153B.champsimtrace.xz
619.lbm_s-2677B.champsimtrace.xz
641.leela_s-800B.champsimtrace.xz
644.nab_s-7928B.champsimtrace.xz

ngober, 2021-04-02:
Could you take a look at #147? I managed to track my issues down to the DRAM model, and I wonder if it would address this deadlock.

closed 2022-05-09 by GinoAC