ChampSim/ChampSim #137: What is the variable for correct prediction?

AminAvan, 2021-03-02:
I need to know what is the variable that stands for correct prediction in ChampSim?

`if (myBranchPredictorPrediction == CORRECTPREDICTION){
 do somthing
} else{
 do something
}`

I need to know "CORRECT-PREDICTION"

Bests

GinoAC, 2021-03-02:
If you're building a branch predictor, the `last_branch_result` function's `taken` argument gives you the correct prediction from the branch seen at address `ip`.

AminAvan, 2021-03-09:
Thanks a lot,
I add successfully TAGE predictor to ChampSim, but the TAGE that I added was the 2006 version of it.
Now, I want to add [TAGE-SC-L in ChampSim](https://www.jilp.org/cbp2016/program.html), but TAGE-SC-L just has a .h file and I confused about where is the .cc file of this predictor, would you please guide me? I appreciate it if you help me to add TAGE-SC-L on the ChampSim, as well
Bests

GinoAC, 2021-03-09:
Unfortunately, I'm unable to help with any implementations right now. I recommend posting any questions about implementing your own mechanisms in the mailing list as this is forum is mainly intended for questions related to ChampSim issues, bugs, and improvements. Feel free to post any questions you have with regards to branching interfacing and operations in ChampSim. 
Regards

closed 2021-03-09 by GinoAC