ChampSim/ChampSim #155: Is it right that 100 cycles take between predicting on a branch and knowing the prediction was correct/wrong?

AminAvan, 2021-04-12:
Hello,

I ran Hashed-perceptron branch prediction by a trace named `631.deepsjeng_s-928B.champsimtrace.xz`, and the general results are:
```Branch Prediction Accuracy: 96.5911%
MPKI: 5.1076
Average ROB Occupancy at Mispredict: 74.5841```

According to the results, this trace consists of 1498335 branches, but some branches take **100 cycles and more even 1000 cycles** between when the branch predictor gets a prediction on a branch until the ChampSim determined that prediction is correct/wrong! The cycle duration is correct?

Kind Regards

sethpugsley, 2021-04-12:
Is the 100 cycles you're observing between the calls to predict_branch() and last_branch_result() for the same branch, or is it between the call to predict_branch() and complete_execution() for the same branch?  If it's the former, then that's very unusual, because those should magically happen on the same cycle in ChampSim.

If it's the latter, then that's to be expected, because a lot has to happen between branch prediction and evaluation of the branch.    There are many things that each instruction needs to do between those points, and the branch may be dependent on many other instructions, which themselves may have high latencies and other dependencies.

AminAvan, 2021-04-13:
Would you please let me know where is the `complete_execution()`? (in main.cc or ooo_cpu, ...)

ngober, 2021-04-13:
`complete_execution()` is a member function of `O3_CPU`. You can find what line it is on with `grep -n src/ooo_cpu.cc`.

AminAvan, 2021-04-13:
Thanks for your kind response

AminAvan, 2021-04-17:
> Is the 100 cycles you're observing between the calls to predict_branch() and last_branch_result() for the same branch, or is it between the call to predict_branch() and complete_execution() for the [...]

how the ChampSim work that the taken/not-taken of the branch is realizing in the same time?

ngober, 2021-05-24:
Realizing the taken/not-taken status of a branch is possible because the information is included in the ChampSim traces. So, the branch predictor is queried to predict the branch and the returned prediction is compared to the trace information.

Regarding the repeated IP, that is not something that it seems like we could answer. It could simply be that the instruction with IP 4222718 is fetched 7839 times.

closed 2022-08-29 by ngober