ChampSim/ChampSim #156: Taking cycle in ChampSim

AminAvan, 2021-04-20:
Hi

In ChampSim, `predict_branch` and `last_branch_result` functions are done in the same cycle(), how can I take for example 5 cycles between those functions?

Bests

sethpugsley, 2021-04-20:
You could put the call to last_branch_result() in another function, such as the retirement function, however I wouldn't recommend doing that unless you're specifically studying the complexities of managing metadata between branch prediction and branch prediction training.

In a real system, the BPU can't know immediately what the real result of a branch is like ChampSim can, so it has to carry a ton of state describing the the CPU at the time the prediction was made all the way through to branch execution, after which time it can train the branch predictor in-order.  You could guarantee the in-order aspect of this by doing it at retirement, but there's still a lot of state you'd have to plumb through every stage of the pipeline.  There's very little realism benefit to doing this.

So that's one way you could do it, but I really don't recommend messing with it unless these complexities are what you're specifically studying.  If you're doing cache studies, prefetching, or even developing new branch predictor algorithms, I wouldn't recommend changing this part of ChampSim.

AminAvan, 2021-04-20:
Thanks for your kind response

I found an approach that without harming the structure of ChampSim, it is possible to add some delays between `predict_branch` and `last_branch_result`, but I need to know a command or line/lines of code that take one cycle to be done then I can use them in the approach, would you please help me to know that kind of commands?

Bests

sethpugsley, 2021-04-21:
On the develop branch of ChampSim, you can use the DelayQueue<> data structure to add latency to work items that you want to just sit around for a fixed number of cycles.  Just make sure that you call .operate() on the queue to advance its notion of time so that work will eventually be ready to pop out of it.  You can look for examples of how this is used like the DISPATCH_BUFFER.

AminAvan, 2021-04-23:
Thanks for your kind response

I cannot find `DelayQueue` yet, would you please tell me what file has `DelayQueue` (dram_controller, ooo_cpu) ?

Bests

ngober, 2021-04-25:
[This](https://github.com/ChampSim/ChampSim/blob/develop/inc/delay_queue.hpp) is the file you're looking for.

closed 2021-05-06 by GinoAC