ChampSim/ChampSim #157: Modifying cycle of ChampSim

AminAvan, 2021-04-24:
Hi

In ChampSim, `predict_branch` and `last_branch_result` functions are done in the same cycle (based on `current_core_cycle[cpu]`), how can I take for example 5 cycles between those functions? i.e., if `predict_branch` is done in `current_core_cycle=2`, the `last_branch_result` function done in `current_core_cycle=5` instead of `current_core_cycle=2`

Bests

ngober, 2021-04-25:
Out of the box, it is not possible to do this in ChampSim. For ease of implementation, the branch predictor operates instantaneously. If you want to do this, you will need to modify the internals of ChampSim, and there are probably a dozen ways to accomplish what you want.

A feature request (implemented in #129) for prefetchers has been to add a cycle_operate hook. If that would be useful, I can look into adding it, but it will be a while before a change like that can make it into the develop branch, let alone master.

AminAvan, 2021-04-25:
Thanks

May I ask another question, How the cycle of ChampSim get increased normally, I mean how it works?

ngober, 2021-04-26:
It is incremented in the main simulation loop, [here](https://github.com/ChampSim/ChampSim/blob/118459cfe900f39b05302c4ecda125d6d25ef88d/src/main.cc#L803).

closed 2021-05-06 by GinoAC