ChampSim/ChampSim #170: There is a problem with the statistical data.

Yujie-Cui, 2021-07-21:
When I just ran ChampSim, the L2 data prefetch method I chose was no. 

But in the final statistics there are still:

**L2C TOTAL     ACCESS:    8668497  HIT:    5826282  MISS:    2842215
L2C LOAD      ACCESS:     106615  HIT:          0  MISS:     106615
L2C RFO       ACCESS:    3978965  HIT:    1847413  MISS:    2131552
L2C PREFETCH  ACCESS:     603967  HIT:          3  MISS:     603964
L2C WRITEBACK ACCESS:    3978950  HIT:    3978866  MISS:         84
L2C PREFETCH  REQUESTED:          0  ISSUED:          0  USEFUL:          1  USELESS:    2841844
L2C AVERAGE MISS LATENCY: 4759.91 cycles**

The above shows that the prefetch issued is 0, but the useful and useless are not 0.

**L1D TOTAL     ACCESS:   35331273  HIT:   32682477  MISS:    2648796
L1D LOAD      ACCESS:   16941674  HIT:   15272962  MISS:    1668712
L1D RFO       ACCESS:   15386509  HIT:   15128980  MISS:     257529
L1D PREFETCH  ACCESS:    3003090  HIT:    2280535  MISS:     722555
L1D WRITEBACK ACCESS:          0  HIT:          0  MISS:          0
L1D PREFETCH  REQUESTED:    4563916  ISSUED:    4563916  USEFUL:     442443  USELESS:     280130
L1D AVERAGE MISS LATENCY: 793.855 cycles**

I set up the L1 data prefetching algorithm. But the sum of useful prefetching and useless prefetching does not equal the total prefetching issued. Even the difference is quite a lot. Why is this?

GinoAC, 2021-08-07:
For the L2, the prefetch information is the number of useful/useless prefetches that the L2 received, including prefetches from the L1.

As for the disparity in the L1D's measured prefetches, the number of **prefetches issued** here is the number of prefetches that were added to the prefetch queue at the L1D. The number of **prefetch accesses** describes the prefetches that were a hit/miss in the L1D and did not match an already existing MSHR or get merged into a previous request. Since a prefetch hit in the L1D by an L1D prefetch does not cause a change in state, a prefetch hit in the L1D is not counted in the useful/useless metrics. Only the L1D prefetch misses, which fills the L1D with a new cache line, are measured for usefulness. 

So you should expected **useful + useless ~= prefetch misses**. Hope this clears things up!

Yujie-Cui, 2021-08-08:
Thank you very much for your answer. But I am not entirely sure that I have understood it correctly.
L1D PREFETCH ACCESS: 3003090 HIT: 2280535 MISS: 722555
L1D PREFETCH REQUESTED: 4563916 ISSUED: 4563916 USEFUL: 442443 USELESS: 280130

Does the above data mean:
The L1D cache had 4,563,916 prefetch requests, and 4,563,916 prefetches were issued. 
Among the 4,563,916 prefetch requests, 3003090 cache accesses have been performed so far, of which 2,280535 have cache hits and 722555 have misses.

A cache miss prefetch needs to bring the cache line to the cache, of which 442443 times are valid prefetches and 280130 are invalid prefetches.

Therefore, the prefetch accuracy rate is 280130 / (442443  + 280130 )
Therefore, the prefetch coverage rate is 1 - 722555/ base_line(MISS) 

Is my understanding correct?
Thanks!

ngober, 2022-01-21:
Your understanding is correct. These are the typical ways to calculate accuracy and coverage.

closed 2022-01-21 by ngober

GinoAC, 2022-05-10:
That is correct!