ChampSim/ChampSim #177: L1 Data Cache Statistics requires correction

AbhishekUoR, 2021-09-30:
Hello everyone,

I have created two counters in file "prefetcher/no.l1d_pref ", the counters are cache_load_accesses and cache_load_misses.

I use "cache_hit" flag to determine the number of misses. For cache_load_access, i just increment counter whenever "void CACHE::l1d_prefetcher_operate" function is called.

I notice that both cache_load_accesses and cache_load_miss do not match with the final statistics file generated "L1D LOAD ACCESS" and "L1D LOAD MISS".

Why do L1 Data cache load accesses and misses do not match with standard counters in the statistics file?

adhar77, 2022-05-30:
could be because you might not be setting your counters to 0 after warmup. Had similar issue in another simulator.

ngober, 2023-03-27:
I'm closing this issue as stale. There are now many tests written that protect against behavior like this. If you still see this behavior on the development branch, please reopen this issue.

closed 2023-03-27 by ngober