ChampSim/ChampSim #183: question and/or bugs on handle_writeback

gxflying, 2021-12-07:
hi , sirs , I am learning about the Champsim source code, and I am confused about the handle_writeback api ：
1.  L316 is on a way where cache_type==L1D, but L316 is a condition that cache_type==LLC  which I think will never be ture ?!
2. L336  call **add_rq** without check the return value,  or make sure that the RQ has enough entries for the requst to be added 
    if the RQ is full , the request will lost ! 
wish your replies,  thanks

https://github.com/ChampSim/ChampSim/blob/55974dda5e2fc4a7d90b68db048e1f4327928a8d/src/cache.cc#L303
https://github.com/ChampSim/ChampSim/blob/55974dda5e2fc4a7d90b68db048e1f4327928a8d/src/cache.cc#L316
https://github.com/ChampSim/ChampSim/blob/55974dda5e2fc4a7d90b68db048e1f4327928a8d/src/cache.cc#L336

ngober, 2021-12-07:
I'll bet the first point was a copy and paste from the adjacent code in `handle_read()` and `handle_prefetch()`. The second probably is a bug.

In both cases, these seem to have been fixed on the [development branch](https://github.com/ChampSim/ChampSim/blob/28663518e2c7c7279573ec3f2c2d6abaf4d9f846/src/cache.cc#L255), so I'm not worried about them.

gxflying, 2021-12-08:
@ngober thanks for your information, 
I look at the **develop** branch and find it much more ahead than the **master** branch
and is the **master** branch still maintained,   whether new features are to be merged into **master** branch ?
if the **develop** branch  is with all the new features maybe I should track it other than the **master** branch
the develop branch overview : This branch is 276 commits ahead, 3 commits behind master.

ngober, 2021-12-08:
The develop branch will be merged soon, hopefully by the end of January. There are a few things moving around behind the scenes, but a merge is certainly overdue.

gxflying, 2021-12-08:
thank you very much

closed 2021-12-08 by gxflying

gxflying, 2021-12-08:
Hi， @ngober I have one more question for the function **handle_writeback**  in the devlop branch
in my opinion, there is no need to treat L1D cache differently when handling the writeback request from upper component
for L1D the request is a write from core, for other level cache the requst is upper level cache, 
acctually I have no idea why the L1D should take an action  other than the common flow ：
if HIT     : modified the block data, bla bla
if MISS  : find a victim,  send request to the lower level cache
can you show me some basic ideas about this ?  
thanks, and wish for your reply
https://github.com/ChampSim/ChampSim/blob/28663518e2c7c7279573ec3f2c2d6abaf4d9f846/src/cache.cc#L94

ngober, 2021-12-08:
ChampSim implements a write-allocate policy. In this policy, an L1 write is different from a writeback because the block must be present before it can be written to. In a write-allocate system, we can support writes of less than a block without bytemasks, and writebacks are simple because the whole of the data is already present so we can evict and replace.

gxflying, 2021-12-09:
thanks, @ngober , it makes sense,  and I have another question that the multi-level cache is inclusive or exlusive ?
when I read the code,  I find it may NI/NE , am I right?
because **if the policy of L1D and L2  are inclusive**,  when L2 has a dirty block to be write back,  it should **invalidate** the  L1D the block, but
in the **handle_fill** function, the evicit dirty flow, it just send a write back request to the lower level cache **without notifies the upper level cache to ivalidate** the same block , which is the basic idea if the inclusive cache
handle_fill->filllike_miss->add_wq into lower_level cache (BUT **without invalidate the upper level cache entry** !)
https://github.com/ChampSim/ChampSim/blob/28663518e2c7c7279573ec3f2c2d6abaf4d9f846/src/cache.cc#L323
**if the policy of L1D and L2  is exclusive**,  when the read request from L1D to L2 is a hit on L2 cache  , the L2 should return data to L1D and invalidate the demanded block in L2,  but I see no such operations  in the handle_read function ?
handle_read->readlike_hit  (return data to upper level cache **without invalidate the block in it self**)
https://github.com/ChampSim/ChampSim/blob/28663518e2c7c7279573ec3f2c2d6abaf4d9f846/src/cache.cc#L208
could please help me to clear it up ?   this is very important to understand the ChampSim system

ngober, 2021-12-10:
ChampSim makes no effort to maintain inclusivity or exclusivity in the cache. This is perhaps a way that the simulator could be extended. I would welcome any pull requests for this.

gxflying, 2021-12-11:
thanks, maybe I can do some works on this topic sometime later