ChampSim/ChampSim #196: Use of NUM_SET and NUM_WAY in instruction prefetchers

ngober, 2022-01-19:
https://github.com/ChampSim/ChampSim/commit/fb87c5daaf3b637363c9e95483999852cf291edd#commitcomment-64121863

@alberto-ros I'm migrating the discussion here so that it has more visibility. CC: @sethpugsley @GinoAC 

The instruction prefetchers are members of O3_CPU and have been since the initial designs. However, when IPC1 was run, the prefetchers had access to the global constants `L1I_NUM_SET`, `L1I_MSHR_SIZE`, etc. Those constants were removed in favor of `CACHE::` member variables. However, this makes `CACHE::NUM_SET` is inaccessible from the instruction prefetchers.

How can we support instruction prefetchers like this? A short-term solution could be to add `O3_CPU::L1I_MSHR_SIZE`, but this isn't a good idea for the future. Alberto suggests making the instruction prefetchers members of `CACHE` instead, but this would break compatibility with old code, and also I'm not sure what to do then about `O3_CPU::prefetcher_branch_operate`.

alberto-ros, 2022-01-19:
Thanks, Nathan.

I do not think that making instruction prefetchers members of CACHE breaks more compatibility than the current version. If I am correct, no prefetcher was using any element of the O3_CPU class in the branch_operate function. So it will be easier to port them in my opinion. I think I will give it a try and let you know :-)

ngober, 2022-02-06:
I had a thought last night. Say we move the instruction prefetching hooks into the `CACHE` class. I think that, if a user wants to continue to use the branching behavior in their prefetcher, it could be possible to just create a dual branch-predictor/prefetcher.

For example:

    mkdir ~/branch_sensitive_prefetcher
    cp branch/perceptron/*.cc ~/branch_sensitive_prefetcher
    cp prefetcher/next_line/*.cc ~/branch_sensitive_prefetcher
    vim ~/branch_sensitive_prefetcher/files.cc

Then, with the configuration file

    {
        "branch_predictor": "~/branch_sensitive_prefetcher"
        "L1I": { "prefetcher": "~/branch_sensitive_prefetcher" }
    }

Then, you can achieve the desired behavior where the branch predictions can trigger prefetches in the cache. Does this seem like a good idea? It would break existing code, especially the code from the last Instruction Prefetching Championship, but it would be a more maintainable solution and would give you the access to `NUM_WAYS` that we're looking for.

alberto-ros, 2022-02-07:
Hi Nathan,

I think that I am missing something here. Do the code of the I-prefetchers use attributes of the O3CPU class? I think that was not allowed in the championship, and they were just using the parameters passed to the branch_operate function. In that case, moving the branch-operate function from the O3CPU class to the CACHE class is very simple in my opinion. The call will still be in the O3CPU code, but it will be called as static_cast<CACHE*>(L1I_bus.lower_level)->branch_operate(...).

Am I missing something here?

ngober, 2022-02-07:
If the branch operation hook is moved to the CACHE class, then all prefetchers, including data prefetchers for which it makes no sense, must define the hook. That isn't a change I want to make. Caches, in general, shouldn't need to specify their behavior on branch instructions.

alberto-ros, 2022-02-08:
I agree that that call should not appear in the prefetcher code. Perhaps, creating that function in a similar way as done now for the differences between the I and D prefetchers could be an option? I mean, adding that case in the config.sh generated code and just define the function (perhaps with an assert(false)?) in the auto-generated code, so the "contestant" does not see that function in the template? Not clear neither to me what is the best option in this case.

ngober, 2022-02-08:
Is [this](https://github.com/ngober/champsim/tree/ipref_to_cache) something like what you're thinking? I put together a hasty draft of what I'm thinking. It still clearly divides prefetchers into either instruction or data prefetchers, but they're all members of the `CACHE` class now.

alberto-ros, 2022-02-09:
To me it seems like a good option. Removing the distinction may be difficult, but now iprefs can easily access the CACHE attributes. 

Just noticed that the instruction prefetcher passes now to cache operate both: uint64_t addr and uint64_t ip. Was this change intended? They should contain the same value and before was just "addr" passed.

ngober, 2022-02-09:
I did that in an attempt to make the differences between instruction and data prefetchers less confusing, but your reaction means I might have been wrong. Consider these two pairs of interfaces:

    // For instructions
    uint32_t CACHE::prefetcher_cache_operate(uint64_t /*ignored*/, uint64_t ip, uint8_t cache_hit, uint8_t type, uint32_t metadata_in);
    // For data
    uint32_t CACHE::prefetcher_cache_operate(uint64_t addr, uint64_t ip, uint8_t cache_hit, uint8_t type, uint32_t metadata_in);

versus

    // For instructions
    uint32_t CACHE::prefetcher_cache_operate(uint64_t ip, uint8_t cache_hit, uint8_t type, uint32_t metadata_in);
    // For data
    uint32_t CACHE::prefetcher_cache_operate(uint64_t addr, uint64_t ip, uint8_t cache_hit, uint8_t type, uint32_t metadata_in);

I can understand how the first one might seem redundant, but I think it produces fewer confusing compilation errors for a first-time user. I think we could just explain in the wiki or README that `(addr == ip)` is an invariant for instruction caches. Plus, this steps towards (without reaching fully) my ideal of having identical interfaces.

alberto-ros, 2022-02-09:
I agree with leaving your new version. I was not sure about if it was a copy-paste mistake or intended :-) 

Adding on the cache_operate of each provided i-prefetcher:  assert(addr == ip); // This is an invariant for an instruction prefetcher
 
Would work for me, and it may be a better place than in a README.

ngober, 2022-03-03:
Closing after the merge of #210.

PR #210 (2022-02-21): Move instruction prefetcher to the cache model
Resolves #196.

This patch moves the instruction prefetcher to become a member of the cache model, rather than the core model. This allows the instruction prefetcher to access information about the cache, such as the number of sets and the number of ways.

Instruction prefetchers must still specify a branch hook, which still distinguishes the instruction prefetchers from the data prefetchers. I think that this constraint can eventually removed, but that is beyond the scope of this patch. One thing I do like about this patch is that it removes a wh [... cut]
closed 2022-03-03 by ngober