ChampSim/ChampSim #211: ITLB accesses

alberto-ros, 2022-02-24:
Hi everyone!

I have been playing with the simulator front-end and found something unexpected regarding ITLB behavior. 

For example, lets assume that 6 instructions to the same page are fetched in cycle 1. The first instruction immediately starts the ITLB look-up and once it is resolved those 6 instructions will be marked as translated.

But let's also assume that in cycle 2 six more instructions to the same page are fetched. They should either initiate a subsequent ITLB access, or being resolved along with the first instruction (fetched in cycle 1). However, none of these things happens, and the second group of instructions waits until the first 6 instructions are moved to decode.

The reason for this is the following if condition: 

 if (itlb_req_end != IFETCH_BUFFER.end() || itlb_req_begin == IFETCH_BUFFER.begin())
    {
        do_translate_fetch(itlb_req_begin, itlb_req_end);
    }

What is the purpose of this condition? It is what I am describing here the expected ITLB behavior?

Also, I noticed that the ITLB actual latency is 2 cycles in case of a hit, while I was expecting to be 1 (as set in the configuration file).

Thanks a lot,
Alberto

alberto-ros, 2022-02-25:
Just more information about this issue. I replace the mentioned if by:

    if (itlb_req_begin != itlb_req_end)

Similarly I did it for the L1I code:

-     if (l1i_req_end != IFETCH_BUFFER.end() || l1i_req_begin == IFETCH_BUFFER.begin())
+    if (l1i_req_begin != l1i_req_end)

And IPC improves, so I think that the baseline has extra stalls that are not necessary.

Any comment about this is welcome.

alberto-ros, 2022-02-25:
I also found why the TLBs have a minimun latency of two cycles.

The key is in when the delay_queue updates the ready items. This is currently done now in the operate function, but that function is executed a cycle after the insertion in the delay_queue. Therefore, the update of delays is done late when the latency is 0.

The solution is simple, just to add a function in delay_queue.hh:

            void update_delay() {
                auto delay_it = std::partition_point(_delays.begin(), _delays.end(), [](long long int x){ return x <= 0; });
                _end_ready = std::next(_buf.begin(), std::distance(_delays.begin(), delay_it));
            }

And call to that function at any push_back:

            void push_back(const T& item) { _buf.push_back(item); _delays.push_back(_latency); update_delay(); }
            void push_back(const T&& item) { _buf.push_back(std::forward<T>(item)); _delays.push_back(_latency); update_delay(); }

            void push_back_ready(const T& item) { _buf.push_back(item); _delays.push_back(0); update_delay(); }
            void push_back_ready(const T&& item) { _buf.push_back(std::forward<T>(item)); _delays.push_back(0); update_delay(); }

Best,
Alberto

alberto-ros, 2022-02-26:
The last fix resolves the current limitation of minimum 2-cycle access time to the cache structures.

ngober, 2022-03-03:
Did #212 resolve this?

closed 2022-05-12 by ngober