ChampSim/ChampSim #243: Multi-core build is failing.

YashikaVerma156, 2022-05-08:
Steps to reproduce: 
1) change "num_cores" in champsim_config.json -->  "num_cores": 2
2) ./config.sh champsim_config.json
3) make

src/core_inst.cc:28:8: error: redefinition of ‘O3_CPU cpu0’
   28 | O3_CPU cpu0(0, 1.0, 32, 8, 16, 64, 32, 32, 352, 128, 72, 6, 6, 6, 128, 4, 2, 2, 5, 1, 1, 1, 0, 0, &cpu0_ITLB, &cpu0_DTLB, &cpu0_L1I, &cpu0_L1D, O3_CPU::bpred_t::bbranchDbimodal, O3_CPU::btb_t::bbtbDbasic_btb, O3_CPU::ipref_t::pprefetcherDno_instr);

src/core_inst.cc:27:8: note: ‘O3_CPU cpu0’ previously declared here
   27 | O3_CPU cpu0(0, 1.0, 32, 8, 16, 64, 32, 32, 352, 128, 72, 6, 6, 6, 128, 4, 2, 2, 5, 1, 1, 1, 0, 0, &cpu0_ITLB, &cpu0_DTLB, &cpu0_L1I, &cpu0_L1D, O3_CPU::bpred_t::bbranchDbimodal, O3_CPU::btb_t::bbtbDbasic_btb, O3_CPU::ipref_t::pprefetcherDno_instr);
      |        ^~~~
make: *** [<builtin>: src/core_inst.o] Error 1

GinoAC, 2022-05-09:
Hi Yashika,

To add another core, you need to also define its configuration in the *.config file. For example, to make another copy of the core, your "ooo_cpu" field should look something like the snippet below. Please let me know if this resolves your issue. 

```
"ooo_cpu": [
        {
            "frequency": 4000,
            "ifetch_buffer_size":64,
            "decode_buffer_size":32,
            "dispatch_buffer_size":32,
            "rob_size": 352,
            "lq_size": 128,
            "sq_size": 72,
            "fetch_width": 6,
            "decode_width": 6,
            "dispatch_width": 6,
            "execute_width": 4,
            "lq_width": 2,
            "sq_width": 2,
            "retire_width": 5,
            "mispredict_penalty": 1,
            "scheduler_size": 128,
            "decode_latency": 1,
            "dispatch_latency": 1,
            "schedule_latency": 0,
            "execute_latency": 0,
            "branch_predictor": "bimodal",
            "btb": "basic_btb"
        },
        {
            "frequency": 4000,
            "ifetch_buffer_size":64,
            "decode_buffer_size":32,
            "dispatch_buffer_size":32,
            "rob_size": 352,
            "lq_size": 128,
            "sq_size": 72,
            "fetch_width": 6,
            "decode_width": 6,
            "dispatch_width": 6,
            "execute_width": 4,
            "lq_width": 2,
            "sq_width": 2,
            "retire_width": 5,
            "mispredict_penalty": 1,
            "scheduler_size": 128,
            "decode_latency": 1,
            "dispatch_latency": 1,
            "schedule_latency": 0,
            "execute_latency": 0,
            "branch_predictor": "bimodal",
            "btb": "basic_btb"
        }
    ]
```

ngober, 2022-05-10:
This is not the intended behavior. Specifying `num_cores` should be sufficient if your intent is to create duplicate cores. I created a patch that would resolve this. I'm currently basing it on develop, but it would apply cleanly to master. It might be worthy of a hotfix, I don't know.

PR #245 (2022-07-13): Fixed bug in duplicating cores
Resolves #243.

This resolves a bug where cores would be named before they were duplicated, breaking multicore configurations unless all cores were fully-specified.