ChampSim/ChampSim #246: Error with multiple traces

agusnt, 2022-05-30:
Hello everyone.

I am trying to execute the SPEC CPU2017 (link to traces: https://hpca23.cse.tamu.edu/champsim-traces/speccpu/index.html) and GAP (link to traces: https://utexas.app.box.com/s/2k54kp8zvrqdfaa8cdhfquvcxwh7yn85/folder/132804598561) traces with the default ChampSim configuration (develop branch), and some traces do not complete execution.

I have issues with the following traces:
- 607.cactuBSSN_s-2421B.champsimtrace.xz
- 607.cactuBSSN_s-3477B.champsimtrace.xz
- 607.cactuBSSN_s-4004B.champsimtrace.xz
- 619.lbm_s-4268B.champsimtrace.xz
- 620.omnetpp_s-874B.champsimtrace.xz
- 621.wrf_s-6673B.champsimtrace.xz
- 628.pop2_s-17B.champsimtrace.xz
- 654.roms_s-1390B.champsimtrace.xz
- bc-5.trace.gz
- pr-5.trace.gz

### Steps to reproduce:
1. ./config.sh champsim_config.json (champsim_config.json pastebin: https://pastebin.com/18Pa6jkL)
2. make
3. bin/champsim --warmup_instructions 50000000 --simulation_instructions 200000000 $dir_to_trace/607.cactuBSSN_s-3477B.champsimtrace.xz
4. Output: https://pastebin.com/EjKDEq7i

### Other useful information:
- Branch: develop
- Commit: 77d96b7d3d27ea20a924747e7c5101b625788e9c

adhar77, 2022-05-30:
I was planning to use champsim for my research work. As there is a bug in running multiple trace and I want to run multiple traces should I use it?

ngober, 2022-05-30:
First, this is "multiple traces" as in there are certain programs and certain configurations that cause a deadlock condition. If you want to run a multicore simulation, then you can probably still do that without any issues.

Second, this is a recently-emerging bug on the development branch that does not affect the stable master branch.

agusnt, 2022-11-09:
Hello again,
I'm fixing the PTW bug, but another bug appeared (debug trace: https://pastebin.com/2hjWatCh)

It seems that instr X (load) is merged in the MSHR with instr Y (PTW) because they have the same physical address.

I think the problem is in how the virtual memory allocates the physical pages to the PTWs:

https://github.com/ChampSim/ChampSim/blob/4ee40fbbe5040be8e447fc116cfa5fb558ef34c8/src/vmem.cc#L45-L60

If I understand the code correctly, the process is as follows:

1. We see if the PTE is already allocated.
2. If it is not allocated (fault flag), we set the next_pte_page as the next page plus the page_size
3. If next_pte_page is NOT equal to the page size we remove that page from the list of available pages.

I think the `if (next_pte_page % PAGE_SIZE)` should be `if (!(next_pte_page % PAGE_SIZE))` because if the mod of the address next_pte_page and PAGE_SIZE is 0 it means that we are out of range of the physical page assigned and therefore we should get a new physical page.

Am I right or am I missing something?

I have tested the patch and it works.

ngober, 2022-11-10:
I think you might be right here. I'm trying to put together a test to isolate this behavior and make sure it doesn't regress, but it's a little tricky to tease out the behavior. In hindsight, it's not a good thing that the implementation of the virtual memory just throws a bunch of pages into a big long container just to pop off it.

ngober, 2023-02-22:
Did #293 resolve this issue? I'm testing on the listed traces and it seems like the bug is fixed.

agusnt, 2023-02-27:
Yes. The patch resolve this issue. I am closing the issue :D

closed 2023-02-27 by agusnt