ChampSim/ChampSim #248: Many errors with multi-core simulation

Hoshikawakaguya, 2022-06-10:
I have issues with multi-core simulation. Sometimes champsim get an error and cannot run anymore, sometimes a deadlock is reported.
Here are some examples.
Steps to reproduce:

1. use this config file [https://pastebin.com/r2YEnbXh](url)
2. make
3. run bin/champsim --warmup_instructions 200000000 --simulation_instructions 500000000 619.lbm_s-2677B.champsimtrace.xz 621.wrf_s-8065B.champsimtrace.xz 623.xalancbmk_s-10B.champsimtrace.xz 628.pop2_s-17B.champsimtrace.xz  
4. get a deadlock

1. use this config file [https://pastebin.com/qVtaej4R](url)
2. make
3. run bin/champsim --warmup_instructions 50000000 --simulation_instructions 200000000 xzFile/605.mcf_s-1152B.champsimtrace.xz xzFile/605.mcf_s-1554B.champsimtrace.xz xzFile/641.leela_s-1083B.champsimtrace.xz xzFile/654.roms_s-1390B.champsimtrace.xz 
4. champsim: src/cache.cc:412: virtual int CACHE::add_rq(PACKET*): Assertion `packet->address != 0' failed.

1. use this config file [https://pastebin.com/RfAxPt3V](url)
2. prefecter is 
    l1d: https://pastebin.com/Ck2d6FPg
    l2c: https://pastebin.com/ibcQSjWg
    llc: https://pastebin.com/rNB72rYP
  It work well with single-core simulation
3. make
4. run bin/champsim --warmup_instructions 200000000 --simulation_instructions 500000000 602.gcc_s-2226B.champsimtrace.xz 619.lbm_s-2677B.champsimtrace.xz 620.omnetpp_s-874B.champsimtrace.xz 649.fotonik3d_s-10881B.champsimtrace.xz 
5. Segmentation fault

ngober, 2022-06-10:
Are you running on the master branch? I'm wondering if the deadlock is similar to #246, since the traces used are similar.

Hoshikawakaguya, 2022-06-10:
I use gitclone in readme, it should be master branch

ngober, 2023-02-24:
I ran the first configuration you provided on the develop branch, and I can't reproduce the deadlock. Are you still seeing it? If not, I'll close this issue.