ChampSim/ChampSim #265: Trace files

amalrasas, 2022-09-01:
Hello I'm trying to find a matrix multiplication trace file is there any in the links you provided? the names are not clear and i'm having problems with installing pin 3.17 on Ubuntu 20 
I executed the following commands 
wget https://software.intel.com/sites/landingpage/pintool/downloads/pin-3.17-98314-g0c048d619-gcc-linux.tar.gz
tar -xzvf pin-3.17-98314-g0c048d619-gcc-linux.tar.gz
cd pin-3.17-98314-g0c048d619-gcc-linux/source/tools/ManualExamples
make
and i got this error after executing make 
line.cpp:15:10: fatal error: pin.H: No such file or directory
   15 | #include "pin.H"
      |          ^~~~~~~
compilation terminated.
make[2]: *** [<builtin>: line] Error 1
make[2]: Target 'line.test' not remade because of errors.
make[2]: Leaving directory '/home/ar/champsim/pin-3.17-98314-g0c048d619-gcc-linux/source/tools/ToolUnitTests'
make[1]: [makefile:53: ToolUnitTests.build] Error 2 (ignored)
make[1]: Leaving directory '/home/ar/champsim/pin-3.17-98314-g0c048d619-gcc-linux/source/tools'

ngober, 2022-09-01:
From the error, it seems that your compiler can't find the PIN headers. Did you follow the instructions provided [here](https://github.com/ChampSim/ChampSim/tree/master/tracer/pin#readme) to build PIN? Make sure your `PIN_ROOT` is exported so that the compiler knows where to look.

amalrasas, 2022-09-02:
I followed the instructions you mentioned and I got a new error : Command 'pin' not found, but there are 16 similar ones.

ngober, 2022-09-02:
It sounds like there's something wrong with your path, which we probably won't be able to diagnose with a back-and-forth over an issue thread. Make sure your `$PATH` or `$PIN_ROOT` variables are set correctly and that you are compiling from the ChampSim pintool directory.

I'm going to close this issue since it is out of scope from ChampSim. It seems like this is a PIN or a runtime issue.

closed 2022-09-02 by ngober