ChampSim/ChampSim #271: MPKI changed from old to new version

djimeneth, 2022-09-09:
Someone emailed me and said that on some traces, including server_039 trace, the MPKI of the hashed perceptron predictor varies significantly between the old (June/July 2021) and latest versions of ChampSim. Maybe look into it?

alberto-ros, 2022-12-07:
It may be not because of the hashed perceptron code, but because the BTB target predictor. Mind that ChampSim offers MPKI of target/condition as a whole. I discovered a performance bug in the BTB and did a pull request regarding this (#258). After that fix, I think that MPKI numbers should be back to normal.

sawansib, 2022-12-09:
Hi, I think this increase in the MPKI was due to an incorrect update of branch type when doing a BTB update. I have created a separate [issue](https://github.com/ChampSim/ChampSim/issues/296) for that. In my test after fixing the bug, the MPKI is the same as the older champsim version.

ngober, 2023-03-08:
I think we've identified what causes this, and discussion is continuing in #296, so I'm closing this issue as a duplicate.

closed 2023-03-08 by ngober