ChampSim/ChampSim #273: How to insert dummy instructions in read queue?

gangwarady, 2022-09-18:
I want to add some dummy instructions between 2 trace instructions. I don't want to do it through the trace, instead I directly want to add these dummy instructions from champsim. On pushing a dummy packet on RQ, I'm getting segmentation fault error. No idea why this error happens. Can anybody help?

GinoAC, 2022-09-19:
By read queue are you referring to the cache model or are you adding the instructions directly into the front-end model?

gangwarady, 2022-09-19:
In cache.cc file and in add_rq(PACKET*) function, while pushing the packet in RQ, I simply change the physical address of the packet or give a dummy physical address and then push_back it in the RQ. This is resulting in segmentation fault error.

GinoAC, 2022-09-20:
I'm not sure if this is your goal, but just to warn you, this will only simulate dummy cache requests not dummy instructions which would interact with the frontend.

I recommend using gdb to narrow down where the segmentation fault is originating. Make sure none of your requests have an address of `0` as this is not handled in the cache model.

ngober, 2022-09-20:
I'll second what Gino said. I wouldn't be surprised if this is a hard problem to solve. The system is currently set up with the assumption that every request is meaningful. There's probably countless places where this is implicit, and tracking them down will be tricky. I would add a ton of assertions to make sure that the dummy instructions do what you want.

closed 2022-11-13 by gangwarady