ChampSim/ChampSim #276: Data overwrite in handle_writeback()

dingqy, 2022-09-21:
Hi,

Sorry for asking this question. I am reading Cache codes and have a question for handle_writeback(). When write request hit the cache block, why does it not update the data in the block? 

if (way < NUM_WAY) // HIT
{
  impl_replacement_update_state(handle_pkt.cpu, set, way, fill_block.address, handle_pkt.ip, 0, handle_pkt.type, 1);

  // COLLECT STATS
  sim_hit[handle_pkt.cpu][handle_pkt.type]++;
  sim_access[handle_pkt.cpu][handle_pkt.type]++;

  // mark dirty
  fill_block.dirty = 1;
}

ngober, 2022-09-21:
What action would you expect that the model would take on a write hit? As it stands, the block is marked as dirty (that is, having been written to), and no further action is taken.

dingqy, 2022-09-21:
Sorry for such stupid question. I am new to simulator.

I also check the retire_rob(). It seems that store data is not considered in this simulation. So it only simulate the write request without any actual data update, right?

I mean whether there are such codes to update the cache block.
fill_block.data = handle_pkt.data;

ngober, 2022-09-21:
You're correct. No actual data is contained in the traces. The simulator only tracks the presence of the blocks. The `PACKET::data` field is used in the translation path to provide the VA/PA mapping.

dingqy, 2022-09-21:
Got it. Thank you!

closed 2022-09-21 by dingqy